iwanders/OBD9141 #4: Arduino DUE functionality?

rocketjosh, 2016-05-10:
First off, thank you very much for this excellent library. I have been using it with an Arduino MEGA without any issues. I can run your "reader" example by only changing these lines to use the Serial1 hardware port:

`#define RX_PIN 19`
`#define TX_PIN 18`

Now I am working on updating my project over to an Arduino DUE, but am having some issues. 

I am using the same interface hardware with both circuits and connecting to the same Serial1 port as the MEGA so that *shouldn't* be an issue (both 3.3V and 5V levels should be fine). I am thinking it has something to do with Serial since I am not even getting the DEBUG to print to the terminal. I am using the Native USB port so I wonder if this has anything to do with it? Any thoughts on what to try next?

Again, thanks so much for your great work!

iwanders, 2016-05-11:
>First off, thank you very much for this excellent library.  I have been using it with an Arduino Mega without any issues.

Thanks for the kind words, I'm really glad there are people using it! Good to hear the library works as-is on the Mega, I've been wanting to test that and I actually ordered a Mega a few weeks ago, you beat me to it :)

> I am using the Native USB port so I wonder if this has anything to do with it? Any thoughts on what to try next?

I haven't got an DUE myself, but I think you are on the right track. The Arduino DUE Getting Started guide has a [section on the serial port](https://www.arduino.cc/en/Guide/ArduinoDue#toc3) which states:
>The USB connector of the Native port is directly connected to the USB host pins of the SAM3X. [...] This port can also be used as a virtual serial port using the "SerialUSB" object in the Arduino programming language.

With this info, I believe that if you want to use the native USB serial port, you would have to change all the `Serial` instances into `SerialUSB`. So instead of `Serial.println("init success")` you would do `SerialUSB.println("init success")`.

Another option would be to use the the other USB connector (they call it the 'Programming Port Serial' in the image), which connects the serial port `Serial` on the `rx` pin 0 and `tx` pin 1 via a serial converter chip. This makes the system similar to the setup on the Mega.

In both situations, be sure to pass the right Serial instance to the OBD9141 bus, `Serial1` should work with the defines for pin 19 and 18, just as you already did with the Mega.

Let me know if this helps you out.

rocketjosh, 2016-05-17:
Thank you for the feedback. In fact, I should have mentioned earlier that I did change every instance of `Serial` to `SerialUSB` using this command: `#define Serial SerialUSB` up on top of the code. It is strange, I would think I should be getting some kind DEBUG info this way. 

I'll keep studying things over but if anything else comes to mind, I'd love to hear it. Thanks again!

iwanders, 2016-05-17:
I would try replacing them manually, so without a `define`. Depending on where your `define` statement is and how the preprocessor works, the define might not have any effect at all. I believe the `-E` flag can be used to do that with gcc, perhaps the Arduino DUE compiler also offers that option? Although it's probably hard to change that in the standard Arduino build environment.

It's worth a shot, but if that doesn't help I'm not really sure what's causing the problems unfortunately. If you happen to find the error, please let us know.

rocketjosh, 2016-05-19:
Thank you again, I will give it a try and let you know what I find out.

iwanders, 2016-05-27:
Several problems manifested itself on the Due, both are caused because the Due behaves differently from the other platforms:
 - Calling `end()` on an Serial port that's not opened yet caused the Due to block indefinitely.
 - Using the pins associated to a Serial port as digital pins and then starting a Serial port again is not possible on the Due, even with a trivial example no data is sent over the serial port once its pins have been used as digital IO pins.

The first one is fixed easily, I added a `Serial.begin()` in the library. The second one requires a hardware modification to work around this issue in the Core; it involves using a second digital pin in order to pull the Tx pin of the Serial port low during the init 5-baud sequence.

The library changes are in 4bd841bf99185719b7f9c61d474f3123bbf14b9d and I added an example for the Due in commit 8ccfc7e55c467660e947fe7f51df58857c3107ff. Please give that a go. I haven't done extensive testing; I mainly tested it with the Due communicating via K-line to a second MCU running the simulator on it and only shortly with a real ECU.

Please let me know if this helps.

rocketjosh, 2016-05-29:
Yes! This works great. I changed the enable pin number since my custom board uses a different pin for this - but that is all! Well done, thank you very much.

closed 2016-08-19 by rocketjosh