iwanders/OBD9141 #7: how to force simulator to perform init with reader

cataurus13, 2017-11-23:
hello iwanders,
thank you for your great work on this library.
i am using it with MC33660 and communications seems to work fine when the init is not done. however although i'm trying for days now, i cant figure out how to perform the init.
i found the intructions in OBD9141sim.h and followed them. but the simulator is not responding to the 5baud signals from the reader. it seams the simulator doesnt even run processInterrupt()
but its checking and running the first if condition from handeInit() over and over again.
i enabled #define OBD9141SIM_SIMULATE_INIT.
but (in setup?) the usage of OBD9141sim car = OBD9141sim(Serial1, RX_PIN, TX_PIN); instead of OBD9141sim car; seems to require the modification of the class too.
thanks for your effort

iwanders, 2017-11-23:
Hey cataurus13,

Thanks for reaching out, always happy to hear someone is using my code. As stated in the [header](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141sim.h#L22) file the simulation of the init is not tested working all that well. However, I'm pretty sure that if stuff is setup correctly the interrupt should fire, so something must be wrong.

```cpp
OBD9141sim car = OBD9141sim(Serial1, RX_PIN, TX_PIN)
```
Is guaranteed to cause problems, because you can't use the constructor like that. I chose to adhere to the Arduino conventions and have an empty constructor with and then initialise with `begin`. Try:
```cpp
OBD9141sim car;
car.begin(Serial1, RX_PIN, TX_PIN);
```
should probably work... See also [here](https://github.com/iwanders/OBD9141/blob/master/examples/simulator/simulator.ino#L22), but besides these generic comments I can do little to help out.

I would prefer to test against a real ECU if possible. If you are trying to create your own init algorithm, also take a look at the [logic captures](https://github.com/iwanders/OBD9141/tree/master/logic) I have recorded from my Kia back when I worked on this. You can open those files with the software from [saleae](https://www.saleae.com), they may provide you with the information you need.

It wouldn't surprise me that even if the interrupt fires the handshake won't succeed. If I remember correctly (It's been a while) it was unreliable, there was some issue with the state machine I never fixed. The simulator was mostly made to test the functionality of the PID retrieval, which does work.

cataurus13, 2017-11-26:
Thanks,

may be the overall problem is that i am not using arduino ide but atom with pio on mac and vs with visualmicro on windows10.

I used salea logic 2 years but on windows10 the drivers for the analyzer do not work so i switched to sigrok and pulseview which does not need drivers.

Your answer was in fact helpful. figured out that attachInterrupt didnt work correctly. I now configure the interrupt with the registers and its all fine.

Could take a while but if you wish i'll share the code with you when its working and may be i could add some logic logs for salea from the cars i tested.

iwanders, 2017-11-29:
Hey,

> may be the overall problem is that i am not using arduino ide but atom with pio on mac and vs with visualmicro on windows10.

It shouldn't, after all, the code on the microcontroller should be somewhat independent of where and how you run the compiler ;)

> Your answer was in fact helpful. figured out that attachInterrupt on pin 8 doesnt work on nano [...]. I now configure the interrupt with the registers and its all fine.

Ah, no, that definitely won't work. Glad you found that! I checked if I should've added a comment in the examples, but pin 8 is only used on the SoftSerial example, that actually requires that pin on Atmega 328 platforms. I've done all development on a Teensy 3, which doesn't have any of these restrictions. Should I add a comment anywhere to prevent other users from running into the same issue!?

>Could take a while but if you wish i'll share the code with you when its working and may be i could add some logic logs for salea from the cars i tested. [..] its working but i still have to clean it up. the elm327 and the OBD9141reader_softserial on nano accepted the init.

Very cool! Really glad you got it working! I presume you had to make some changes to the state machine of the simulator? But yeah, would be nice to have the improved simulator in the library, feel free to create a PR or comment with your code here. I'll leave this issue open for now.

cataurus13, 2018-05-26:
hey iwanders,
because of lack of time, a dead Macbook and memory shortage of the Nano, I switched to Teensy. thanks for this great work...

closed 2018-05-26 by cataurus13

iwanders, 2018-05-27:
@cataurus13 , thanks for the update, glad it's working now! :+1: