iwanders/OBD9141 #16: Custom request, and recv full data message

Mebius84, 2019-06-12:
Good day! I need to send my own requests through Iso9141 at a speed of 4800 baud, is it possible with the help of your library?

Set the request header and send / receive data. my car works on its own protocol Subaru select monitor (ssm2), it is physically identical to ISO9141, but the method of obtaining data is different, instead of PID, I need to request reading from the address and parse the answer myself.

iwanders, 2019-06-13:
Hey, what you need should all be possible.

You should be able to change the baud rate [here](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.h#L22).

> the method of obtaining data is different, instead of PID, I need to request reading from the address and parse the answer myself.

You can probably get this to work by calling the `request()` method yourself, take a look at how [`getPID`](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.cpp#L278-L289) works, you can access the full data from the response through the [`readBuffer`](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.h#L182) method. Let me know if this works for you, if you needed to modify more things to add flexibility to support your requirements feel free to send a PR this way. If you don't, I'd still like to hear whether you got it to work, good luck :)

Mebius84, 2019-06-16:
today i try to make request to my ecu, but always first byte in request is 0xC8 but i set 0x80,
and i have wrong checksum and bad request.
i try set buf[0] = 0x80 before calc checksum in "request(void* request, uint8_t request_len)" func and get good CS and right request

my request:
uint8_t ssmMessage[12] = {0x80, 0x10, 0xF0, 0x08, 0xA8, 0x00, 0x00, 0x00, 0x08, 0x00, 0x00, 0x1C};
in debug mode see:
W: 0xC8 0x10 0xF0 0x08 0xA8 0x00 0x00 0x00 0x08 0x00 0x00 0x1C 0x9C