iwanders/OBD9141 #17: Support for different control modules?

isage, 2019-08-22:
This library uses hardcoded address for ECU.
However VAG autos use non-standard ECU address (0x01) and have many other control modules on other addresses.

iwanders, 2019-08-24:
Hey,

Cool, I didn't know that. I've never needed to talk to any other addresses.

I'd be open to a PR to change the address to which we are talking. Something like;

```
void setDestinationAddress(uint8_t address);
uint8_t getDestinationAddress() const;
```

Then we can use the `getDestinationAddress()` call internally whenever we create request. We would default the value to the current address to make sure it still 'works out of the box' in most cases.

domnulvlad, 2024-05-24:
If anyone else is interested in this topic, I have some more information.

In VAG cars with a K-line, address 0x33 will always refer to Universal OBDII diagnostics, commonly ISO9141 and sometimes KWP2000.

All other addresses use a proprietary protocol. Older cars use the in-house "KWP1281" protocol, for which I have written a [library](https://github.com/domnulvlad/KLineKWP1281Lib). Newer modules use KWP2000, but I cannot comment on whether or not it is the exact same protocol as "universal" KWP2000.

In any case the first difference is that KWP1281 modules immediately send their part number, serial number etc. immediately after the 0x55, KW1 and KW2, and keep-alive is achieved by a constant back-and-forth of "acknowledge" messages. Every byte must receive its complement (inversion) from the other party. Real-time data is read from these modules using a "block" system, where a block may contain between 0 and 4 measurements, at any location within the maximum 255 block limit. Then each measurement from a block is described by a "formula" byte, which uses a look-up table to calculate a physical value. There is also a table of enumerated strings that correspond to a specific formula (you can find all of them in my library, extracted from a tester). The block stuff is also true for KWP2000 on modules other than 0x33, but I think the maximum block is 127 and to access anything after it, you actually access block-0x80 and process the response from the 5th measurement onward, if it exists. The fault codes are also different; they are known by a 5-digit value which is just the received code shown as an integer left-padded with 0s. It is a 2-byte value, whose description is retrieved by looking the code up in a table of strings (also found in my library). There is also a status/elaboration for each fault, I also have these strings, but they are different for KWP2000 which I haven't investigated too thoroughly.

All in all, it would not benefit your library to add the possibility of connecting to modules other than 0x33, since they won't use ISO9141.

iwanders, 2024-05-24:
@domnulvlad , thanks for chiming in with these insights! I've added the resolved label and will mark it as closed. If anyone searches for this functionality they should be able to find this issue and their way to your library. Thanks for sharing.

closed 2024-05-24 by iwanders