iwanders/OBD9141 #18: TLE 7259G

kipergrof, 2019-09-09:
Hello. Do you think this chip will work with your library?

https://lomex.hu/pdf/(inf)_tle7259g.pdf

iwanders, 2019-09-14:
It does appear to be a generic LIN driver IC. I see no reason why it wouldn't work.

If you do test it, please report your findings here for others to read :)

kipergrof, 2019-09-14:
I tested it but it doesn't work. I don't know what the problem is. i use esp32, arduino IDE, KWP fast, and your schematic. I was watching Kline with multimeter and saw that there is a voltage drop when I send data.

do you have any ideas?

here is the TLE7259G description: http://www.mouser.com/ds/2/196/Infineon-TLE7259G-DS-v02_01-en-785783.pdf

iwanders, 2019-09-14:
Hmm, looks like the EN pin works in the same way as on the chips I have used. So just pulling it high should enable the chip.

> I was watching Kline with multimeter and saw that there is a voltage drop when I send data.

This implies things are at least working more or less. If you perform a normal init the bus is pulled low and high several times, during [these](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.cpp#L363-L368) the K line should be 0v and battery voltage. If that works, and the K line is at battery voltage when you don't send anything on the serial port I expect your hardware works.

Then the problem likely originates in the communication. Are you sure the car you are trying to communicate with uses KWP? Does it maybe require a different address like in #17? Do you have a working ODB dongle that can communicate with the ECU that you can perhaps sniff to see what the Arduino does differently?

kipergrof, 2019-09-16:
I think esp32 serial port is the problem. Tomorrow I'll meet a man who can help me. If there is any result I will indicate here.

kipergrof, 2019-09-28:
Sorry to answer now, but i was not successful with the project. I managed to get an Arduino UNO and that doesn't work with initialization. I suspect the defective IC or the circuit is not good what I built. Can you look at the datasheet and send me what circuit I need to build exactly?

iwanders, 2019-09-28:
> Sorry to answer now, but i was not successful with the project. I managed to get an Arduino UNO and that doesn't work with initialization. 

I'm sorry to hear that.

> I suspect the defective IC or the circuit is not good what I built. Can you look at the datasheet and send me what circuit I need to build exactly?

See section 3 starting on page 21, the example application diagram on page 22, figure 11 seems to be what you want. The section below the diagram describe exactly how to choose values for the typical applications. I'd start with 2k pullups on Rx and Tx, 220 pF on  LIN (as per fig 22). You can probably get away with leaving WK unconnected, but in doubt use 2k to pullup to Vbat. You may need a pullup on the LIN line towards vbat as well, that depends on your car, try something like 1k on that.
If that doesn't work, I recommend you to use one of the other chips that can be used, some chips that people have had success with can be found in the readme.

> I have my original elm327. the device indicates that the car is using KWP fast initialization.

There's still multiple ways to do the fast initialisation, in #17 I learned that some cars use different addresses for devices on the bus. If you can a logic analyzer capture of the LIN bus can help debug these issues, as we can compare what the ELM327 sends to what your hardware and microcontroller sends.

kipergrof, 2019-09-30:
Did you think of this? Sorry electronics is not my strength :)

[Schematic_test1_Sheet_1_20190930171236.pdf](https://github.com/iwanders/OBD9141/files/3671891/Schematic_test1_Sheet_1_20190930171236.pdf)

iwanders, 2019-10-01:
Yeah, give that a go.

Consider adding the two optional ones I commented about in the previous comment if it doesn't work:
> You can probably get away with leaving WK unconnected, but in doubt use 2k to pullup to Vbat. You may need a pullup on the LIN line towards vbat as well, that depends on your car, try something like 1k on that.

kipergrof, 2019-10-04:
New Schematic.  I'll be testing it over the weekend.

[image]

iwanders, 2019-10-04:
That's missing a pullup on Tx though, not sure if that's necessary, bit compared to that and the previous pdf you shared it's a difference.

kipergrof, 2019-10-05:
I built my circuit. I first tested it without a car. when Tx is high k-line 5V, when TX is low k-line 0V. I think that's good. later I tried the car did not work. another test without a car: when Tx is high then 3.1V, when Tx is low k-line 2.8V. i guess ICs get stuck in sleep or stand by mode. 
what do you think about this?

iwanders, 2019-10-05:
Can be sleep, but if it is sleep I'd expect no difference in voltage on the K line. Does the K line have sufficient pullup?

kipergrof, 2019-10-12:
Report: I bought a new L9637 IC. This is how all functions work. Either the TLE 7259G IC I received is faulty or it is not capable of handling the iso 9141 / iso 14320. Thank you for your help. Have a nice day!

closed 2019-10-12 by kipergrof

iwanders, 2019-10-12:
Thanks for reporting this resolution. Good choice to try with an alternative chip. Happy it works now :+1: