iwanders/OBD9141 #30: about kline

rubensx, 2020-04-21:
Hi Ivor

I'm trying to comunicate k-line data sending ECT temperature to a cluster from a honda s2000 who has a k-line communication, maybe your project can help me? I know this can READ a k-line, but can send k-line messages too? 

thank you

arezaie14, 2020-04-21:
Hello, to read a kline message you need to send a request then get a response.
Yes, this library can read and send a message too. it supports both kwp2000 and iso9141, but this library has a problem with its kwp2000 slow init, if it needed I can help.

rubensx, 2020-04-21:
Hi arezaire!! Thank you for your help! 

My trouble here is, I’m with a s2000 ‘07 honda cluster and the ECT are over k-line (Can protocol). I don’t have the car, only the cluster, and trying to build a analogo to k-line ect converter and sent to the cluster.

I’m trying a arduino can shield and read a message from the cluster or send random message but don’t work!

Can you help?

arezaie14, 2020-04-21:
Kline is on serial protocol not on can. How do you want to run it?

rubensx, 2020-04-21:
Engine cooling temperature default messages from k-line. If I can broadcast these messages to the cluster, I can use this! 

L9637d smd soic8 are recomended to use kline, but still stuck here

arezaie14, 2020-04-21:
See Kline is on Serial Protocol you said you are using CAN bus, yes?

rubensx, 2020-04-21:
Yes, trying to see messages I’m use mcp2515 can shield but not effective.
My intention was to send messages in hex directly to the cluster 

But nothing happens, so I get a L9637d so create these messages, but still Need the kline protocol to discover the right ECT messages to send

It’s a completely blind project (hard to solve)

rubensx, 2020-04-21:
[image]

rubensx, 2020-04-21:
https://forum.autosportlabs.com/viewtopic.php?p=27056&sid=7970aea37d1be38840baf171bc34fc9c

Some information about

rubensx, 2020-04-21:
At last the receptor of this communication on the cluster is this 
[image]

arezaie14, 2020-04-22:
But does it support K-line protocol? If yes then you can make a PCB for l9637d and trying to connect to the device, K-line is more different than CAN bus protocol, it need to Serial port on baud 10400 bps.
then it needs your device headers too, to connect to the device. this library only supports global standard PIDs

roma6868, 2020-04-22:
Hello Ivor,
I have a Peugeot 406 coupe from 2000, I use a Peugeot diagnostic box to communicate with the BSI  , the communication is ISO 9141-2 10400 bps 8-N-1 on the pin 11 of the OBD socket.

And what your code allows to send other than PID ?
I have to send  0x81 0x00 0xF1 0x3E 0X80 for the BSI to respond.

arezaie14, 2020-04-22:
> Hello Ivor,
> I have a Peugeot 406 coupe from 2000, I use a Peugeot diagnostic box to communicate with the BSI , the communication is ISO 9141-2 10400 bps 8-N-1 on the pin 11 of the OBD socket.
>
> And [...]

Hello, Are you initializing it at first? and do you have its header?

rubensx, 2020-04-22:
> But does it support K-line protocol? If yes then you can make a PCB for l9637d and trying to connect to the device, K-line is more different than CAN bus protocol, it need to Serial port on baud [...]

This is the problem.

First I need to Identify the ECT PIDs (the link below makes a far idea of the PID 300) and a format to message the cluster

I believe, this standard message can drive the gauge but, still not clear to me

You say I can do that with a standard arduino serial port out? Or still I need this l9637d?

Maybe the cluster need a initialize too? 

And yes, I come tho this project to get some idea of this messages and a sketch to send this

arezaie14, 2020-04-22:
l9637 is just for changing voltage range from 5V to 12V and you can use the transistor for that there are some examples on the internet, for example: 
[this image](https://blog.perquin.com/prj/obdii/obdii_avr.gif)
You need to change your voltage of the tx of MCU to send a command to ecu.
you will connect tx and RX of usart pins of MCU to l9637d.

Kline protocols are different from the CAN bus. if your Ecu or device support CAN, it doesn't mean that it can support K-line, You need to check it if support.

if it supports you need to know its headers to communicate with that.

> This is the problem.
>
> First I need to Identify the ECT PIDs (the link below makes a far idea of the PID 300) and a format to message the cluster
>
> I believe, this standard message can drive the gauge [...]

roma6868, 2020-04-22:
> 
> 
> > Hello Ivor,
> > I have a Peugeot 406 coupe from 2000, I use a Peugeot diagnostic box to communicate with the BSI , the communication is ISO 9141-2 10400 bps 8-N-1 on the pin 11 of the OBD socket.
> > And what your code allows to send other than PID ?
> > I have to send 0x81 0x00 0xF1 0x3E 0X80 for the BSI to respond.
> 
> Hello, Are you initializing it at first? and do you have its header? You can send any iso 9141 requests to devices with this library

Here is a tram a ping that is done between the diagnostic tool and the BSI :
[image]

https://github.com/roma6868/Peugeot_406_coupe_automatisation/blob/master/SCAN_OBD_PP2000/SCAN_OBD_PP2000.pdf

arezaie14, 2020-04-22:
> Here is a tram a ping that is done between the diagnostic tool and the BSI :
> [image]
>
> https://github.com/roma6868/Peugeot_406_coupe_automatisation/blob/master/SCAN_OBD_PP2000/SCAN_OBD_PP2000.pdf

Yes, you can use this library, you can send any request you want.

iwanders, 2020-04-23:
@arezaie14, thanks for helping out, really appreciated, I only have experience with this library and two cars, so additional insights from others are really welcome :thumbsup:.

@rubensx, to add to what @arezaie14 said;

> At last the receptor of this communication on the cluster is this (IC with 33290 on it)

Thats a a K line interface IC, so yes, this library can likely be used. 

>https://forum.autosportlabs.com/viewtopic.php?p=27056&sid=7970aea37d1be38840baf171bc34fc9c
>
> Some information about

That shows you some of the PID values, like the brake pedal is ID 106, throttle 170... What's missing is how to initialise the bus, but maybe that's not necessary?

---

@roma6868, you should've really made a new issue with this question, because it's different device and use case than the previous discussion.

> And what your code allows to send other than PID ?

Did you read the header file? Did you look in the cpp file to see how the library retrieves PID values? With a bit of reading the code in this library I think you would've been able to figure this out.

@arezaie14 is correct in stating you can send whatever you want. Take a look at how the [getPID](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.cpp#L278-L289) call is implemented. The header file also describes the [raw request](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.h#L136-L160) methods that can be used to send something onto the bus.

arezaie14, 2020-04-23:
@iwanders It's my pleasure and thank you for your best library, I'm using it much more.

roma6868, 2020-04-23:
> 
> 
> @arezaie14, thanks for helping out, really appreciated, I only have experience with this library and two cars, so additional insights from others are really welcome +1.
> 
> @rubensx, to add to what @arezaie14 said;
> 
> > At last the receptor of this communication on the cluster is this (IC with 33290 on it)
> 
> Thats a a K line interface IC, so yes, this library can likely be used.
> 
> > https://forum.autosportlabs.com/viewtopic.php?p=27056&sid=7970aea37d1be38840baf171bc34fc9c
> > Some information about
> 
> That shows you some of the PID values, like the brake pedal is ID 106, throttle 170... What's missing is how to initialise the bus, but maybe that's not necessary?
> 
> @roma6868, you should've really made a new issue with this question, because it's different device and use case than the previous discussion.
> 
> > And what your code allows to send other than PID ?
> 
> Did you read the header file? Did you look in the cpp file to see how the library retrieves PID values? With a bit of reading the code in this library I think you would've been able to figure this out.
> 
> @arezaie14 is correct in stating you can send whatever you want. Take a look at how the [getPID](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.cpp#L278-L289) call is implemented. The header file also describes the [raw request](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.h#L136-L160) methods that can be used to send something onto the bus.

I start in programming I don’t understand everything in the code of your library

rubensx, 2020-04-23:
> @arezaie14, thanks for helping out, really appreciated, I only have experience with this library and two cars, so additional insights from others are really welcome 👍.
> 
> @rubensx, to add to what @arezaie14 said;
> 
> > At last the receptor of this communication on the cluster is this (IC with 33290 on it)
> 
> Thats a a K line interface IC, so yes, this library can likely be used.
> 
> > https://forum.autosportlabs.com/viewtopic.php?p=27056&sid=7970aea37d1be38840baf171bc34fc9c
> > Some information about
> 
> That shows you some of the PID values, like the brake pedal is ID 106, throttle 170... What's missing is how to initialise the bus, but maybe that's not necessary?
> 
> @roma6868, you should've really made a new issue with this question, because it's different device and use case than the previous discussion.
> 
> > And what your code allows to send other than PID ?
> 
> Did you read the header file? Did you look in the cpp file to see how the library retrieves PID values? With a bit of reading the code in this library I think you would've been able to figure this out.
> 
> @arezaie14 is correct in stating you can send whatever you want. Take a look at how the [getPID](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.cpp#L278-L289) call is implemented. The header file also describes the [raw request](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.h#L136-L160) methods that can be used to send something onto the bus.

Yes my problem here is:

I’m using a honda S2000 cluster at 2007 year, it means, all the cluster is analog or I/O (sometimes a digital pulse and arduino can solve tachometer, speed, signals) but can’t solve the engine coolant temperature.

This temperature is provided by a delta calc by ECU:

The ambient temp + oil temp + ECT and the result is broadcasting to cluster...

In some research I found the information:

ECT is Engine Coolant Temp (ID = 300 byte 1), Ambient Temp (ID = 300 byte 2)

It’s founded in a S2000 group of a dataloggers and a specific hardware.

I don’t understand so much how I can create a arduino broadcast of these parameters to test in cluster.

Orher issue is, I don’t have a s2000 ecu or more information about to try this!

This problem affect all honda enthusiasts who want to put a s2000 2006+ cluster in old civics

I had some parts here:

Arduinos: nano, onde and mega
Can shields (3)
At now the kline interpreter to test

If you can help with somenthing to light my path I appreciate so much!

Thank you again

rubensx, 2020-04-23:
hey I guess I buy the l9637d for nothing... 

If I use 33290 IC for de cluster linking the serial RX and TX directly from arduino I cansend K-line messages directly? (I'm using a can shield for a while guessing the can shield speaking with the cluster...

Or still need l9637d and attached in arduino and them to 33290 IC? (arduino cannot connect directly?)

rubensx, 2020-04-26:
Ok I'm back and creating the system...
Goals:

- Create a arduino k-line message sender (cluster baybe not responde nothing to ECU and since I don't have a ECU, maybe I don't need "read" nothing).
- using the information about s2000 (ECT is Engine Coolant Temp (ID = 300 byte 1), Ambient Temp (ID = 300 byte 2)) I'll test if the temp gauge respond...
- at laste create a digital translator from analog temp to K-line messages to the gauge

Questions:

1- some can help me to translate this ID 300 byte 1 into a hex message to the cluster?
2- consider the image above, without a bluetooth part, maybe safe to conect the cluster?
3- I've doubts about baundrate... 10.400, if I use this I can't path the serial messages by the serial monitor from arduino right?

thank you again

rubensx, 2020-04-26:
[https://user-images.githubusercontent.com/36166688/48059537-b9f76e00-e1b9-11e8-827d-115467edb4ef.png](url)

corteXnzt, 2020-10-16:
Hello @rubensx , I am on the same dilemma right now.  How was your end result?

I'm using teensy 3.2 and this circuit to interface with the k-line.
https://www.instructables.com/Low-Cost-OBD2-Communications-on-K-line-ISO-9141-2-/

I also tried tapping directly to the UART of the S2000 cluster connected on the K Line transceiver but reads nothing.

From what  I understand (CMIIW) those ID's are applicable to the ECU/ECM unit via the OBD port. But the cluster is not a master like the ECU/ECM so you can't actually poll anything. 

S2000 cluster k line<----OBD PORT -->ECU they are inter connected on one bus.

So what I expect is the cluster should request data from the ECU just like in the OBD port. But can't read anything going out from the cluster k line port or even the UART port assuming my k line decoding circuit is not working.

What I'm thinking right now is there is like a communication start bit(s) that comes from the ECU before this cluster requests data.

zakirsheik, 2020-12-29:
@rubensx This is a interesting project, I am trying a similar project on a Suzuki Vehicle, trying to drive a Tachometer using K-Line,
I am currently using the OBDSimulator example but facing some issues, will post the updates once I make some progress. Are you trying to use the simulator sketch in the library to drive the S2000 cluster ?

dajcshouse, 2025-08-28:
Hi, I finally found a new development for the s2000 speedometer here. 2006+. I got it too and it works 95%, like you I can't get the temperature to work. So, have you had any success?

dajcshouse, 2026-01-03:
https://chatgpt.com/share/6959173d-250c-8002-92f7-7bd42e33c811
Sorry, but it's in Hungarian, that's I asked chatgpt, but I haven't done any programming yet, but chatgpt wrote some code, if someone could do it and test it, that would be great.