iwanders/OBD9141 #31: Slow init simulator code

bonadio, 2020-09-21:
Hello @iwanders first thanks for this great library I am learning a lot with it.

I am developing some code that does Slow Init iso9141-2 and I am using your simulator code. My question is on https://github.com/iwanders/OBD9141/blob/master/src/OBD9141sim.cpp line 213 you compare "this->init_hold_value == 21" but shouldn't that be "this->init_hold_value == 51" because the lib sends 0x33 as init sequence?

Thank you

iwanders, 2020-09-21:
Hmm, could be, I thought the 21 was because of the slow 5 baud init [here](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.cpp#L363-L370). I thought that was read as `21`, but it is a long time since I used the simulator. It's gotten zero attention after I had working hardware and working initialisation code. I wouldn't really use the simulator to help you develop your own slow init to be honest. It's not created for that purpose, it was mostly created such that I could check my own PID reading code, at some point the simulator did behave identical to my car (including the init), but like I said, detecting the 5 baud init may not be 100% reliable.

In the example where I used it [here](/iwanders/OBD9141/blob/master/examples/simulator/simulator.ino#L23-L24) I actually force the initialisation to be true, this allows you to bypass the initialisation and still allows testing hardware (and reading PID values) as if it is connected to a car. The simulator does work well for acting like a car when it comes to reading PID values.

bonadio, 2020-09-21:
I could not try the code yet, still working on the transceiver but looked again and seems it should be 204 = binary 11001100.

No problem, thanks for the advice, I will use the simulator code it with care. I will post my results once it works.

Thanks very much

bonadio, 2020-09-22:
Hello @iwanders your code is correct the value is 21 I just tried. For some reason I had bit banging in my head and I could not understand how sending 11001100 could be 21, but your code is just counting the number of times the port changes so yes I get 010101 and it is 21.

closed 2020-09-22 by bonadio

iwanders, 2020-09-23:
Ah, glad you got it figured out @bonadio! Thanks for the update :+1: