iwanders/OBD9141 #34: V1 other than V2!

tistructor, 2022-06-18:
Hi,
I was trying to communicate with ODB2 with my 206 1.1cc auto peugeot which should have the KWB2000 slow protocol.
I built the optoisolated hardware interface and it works properly with perfect signal ascent and descent fronts.
I started testing examples of read_softserial.ino library and modified readerKWP.ino for AltSoftSerial.
The first example with debug enabled has this log :
"
Looping
Before magic 5 baud.
Before setting port.
After setting port.
First read is: 85
read v1: 233
read v2: 143
v1: 233
v2: 143
init_success: 0
"
The problem is that the numbers V1 and V2 are different and initialization fails.

With the second example it fails without log.

I don't understand why V1 values other than V2.

Some suggestions?

tistructor, 2022-06-18:
I did it ;-)
The protocol is KWP2000 slow.
With the initKWP function, initialization was unsuccessful.
 So I modified it this way :

```
bool OBD9141::initKWP(){
    use_kwp_ = true;
    // this function performs the ISO9141 5-baud 'slow' init.
    this->set_port(false); // disable the port.
    this->kline(true);
    
    delay(OBD9141_INIT_IDLE_BUS_BEFORE); // no traffic on bus for 3 seconds.
    OBD9141println("Before magic 5 baud.");
    // next, send the startup 5 baud init..  
    this->kline(false); //digitalWrite(7,LOW); 
    delay(200); // start
        
    this->kline(true); //digitalWrite(7,HIGH); 
    delay(400);  // first two bits
    
    this->kline(false); //digitalWrite(7,LOW);
    delay(400); // second pair
    
    this->kline(true); //digitalWrite(7,HIGH);
    delay(400);  // third pair
    
    this->kline(false); //digitalWrite(7,LOW);
    delay(400); // last pair
    
    this->kline(true); //digitalWrite(7,HIGH);
    delay(200);  // stop bit
    
    // this last 200 ms delay could also be put in the setTimeout below.
    // But the spec says we have a stop bit.

    // done, from now on it the bus can be treated ad a 10400 baud serial port.

    OBD9141println("Before setting port.");
    this->set_port(true);
    OBD9141println("After setting port.");
    uint8_t mess[1];

    this->serial->setTimeout(300+200);
    // wait should be between 20 and 300 ms long

    // read first value into buffer, should be 0x55
    if (this->serial->readBytes(mess, 1)){
        OBD9141print("First read is: "); OBD9141println(mess[0]);
        if (mess[0] != 0x55){
            return false;
        }
    } else {
        OBD9141println("Timeout on read 0x55.");
        return false;
    }
    // we get here after we have passed receiving the first 0x55 from ecu.

    this->serial->setTimeout(20); // w2 and w3 are pauses between 5 and 20 ms

    uint8_t v1=0, v2=0; // sent by car:  (either 0x08 or 0x94)

    // read v1
    if (!this->serial->readBytes(mess, 1)){
        OBD9141println("Timeout on read v1.");
        return false;
    } else {
        v1 = mess[0];
        OBD9141print("read v1: "); OBD9141println(v1);
    }
[... 153 lines cut ...]
init_success:1
```
With car on but stopped:
```
Result 0x11 (throttle): 63
Rem: 7
ret_len: 8
R: 132 241 16 65 12 22 108 84 0 
calc cs: 84
buf cs: 84
ok
7
Result 0x0C (RPM): 1435
Rem: 6
ret_len: 7
R: 131 241 16 65 13 0 210 0 
calc cs: 210
buf cs: 210
ok
6
Result 0x0D (speed): 0
```
Optoisolated interface scheme :
[image]
[image]

iwanders, 2022-06-18:
Hey,

Hmm, I've never seen `v1` and `v2` not being identical, I think the spec says they should be... The values you get (233 and 143) as not even bitwise inverts of each other or something that could perhaps make sense... What is good to see is that your updated initialization code still sends `~v2` and then we get the `0xCC` back from the ECU that confirms the initialization succeeded. So you still get the confirmation the ECU is turned on and responding to your requests.

I wonder if the slow part of your initKWP function is somehow just acting as a wakeup signal or something and the end of the KWP init still takes care of the actual initialization :thinking: Regardless, it is working, that's what matters! 

I'm glad you got it working :) Thanks for reporting the problem you encountered and the resolution.

tistructor, 2022-06-19:
I left this information to initialize odb2 kwp2000 :
https://m0agx.eu/2018/01/02/reading-obd2-data-without-elm327-part-2-k-line/

iwanders, 2022-10-03:
I never got back to this, nor did I read the link you sent. But when answering on #41, I realised that the 233 and 143 looked familiar and looked this issue back up. 

I've just filed https://github.com/iwanders/OBD9141/pull/42 which implements the same discoveries we had in #41 as you had to make this work.