iwanders/OBD9141 #35: ESP32 support

hulktech, 2022-07-15:
Hello, i'm trying to use your library on esp32.I use obd.begin(Serial2, RX_PIN, TX_PIN) and pins: rx 13 and tx :32. But Serial2 default pins are 16 and 17.Even if i use the default hardware pins the rx pin is not pull up even if i mod the cpp file to INPUT_PULLUP on rx pin. 

On the other side AltsoftSerial is not compatible with esp32.The alternative option for esp32 is SoftwareSerial ,
So is it possible to add software serial option for eps32 ? 
Thank you in advance!

iwanders, 2022-07-15:
Interesting, so some pointers and ideas...

> Even if i use the default hardware pins the rx pin is not pull up even
it could be that the pullup of the ESP32 is weak, on the rx pin you should just be able to use an external pullup to make it stronger.

During the initialisation, the serial port is actually disabled and the k-line is toggled like a gpio [here](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.cpp#L355-L370). I'm not sure how the ESP32 works, and whether disabling the serial port actually allows the pins to be used as true GPIO pins. The arduino DUE for example also did not behave correctly when I originally wrote this library and I ended up [with these lines](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.cpp#L21-L38) just to toggle the pin functionality between serial and gpio mode. Maybe the ESP32 needs something similar.

Another thing you may run into is that the baud rate is a pretty [odd number](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.h#L22) that may not be handled well by the esp32?

> So is it possible to add software serial option for eps32

SoftwareSerial, or any other serial should work, if you modify the defines [here](https://github.com/iwanders/OBD9141/blob/f1a913b2d7945e2c47b744fb25903ce844258921/src/OBD9141.h#L90-L98) you should be able to easily plug in any other object that supports the required methods. May be easiest to just copy the OBD9141.h and OBD9141.cpp files into your project directory to ensure they're compiled with your changes.

> Also debug option not working on eps32 (cant compile).

Odd, what's the error? It's just doing printing and it should all be arduino-compatible printing.

hulktech, 2022-07-16:
Thank you iwanders for your fast reply! 

Before i explain what changes i made, i have to note that i use arduino core 1.0.6 to compile my sketchs not ESP-IDF. 

So i modified the defines in OBD9141.h ,add : 
#ifdef  OBD9141_USE_ESP32
#include <SoftwareSerial.h>
#define OBD_SERIAL_DATA_TYPE SoftwareSerial

i modified also the OBD9141.cpp file  : 
pinMode(this->rx_pin, INPUT_PULLUP);  (Input Pullup)

Then check while init process,  i had digital output on tx (0x33) and echo on rx.So pull up on rx worked fine.
The next step is to check on car.The crazy thing is that my car is peugeot 206 1.1cc like the one tistructor issue with (KWP Slow init with v1 != v2)   :)))  So i'll check with a car that use ISO9141 Protocol to be sure.

Last thing about debug:
when i uncomment  #define OBD9141_DEBUG i get this errors (sorry for the long list) : 

C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp: In member function 'uint8_t OBD9141::request(void*, uint8_t)':
C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:179:65: error: no matching function for call to 'min(uint8_t&, int)'
     for (uint8_t i=0; i < min(answer_length, OBD9141_BUFFER_SIZE); i++){
                                                                 ^
In file included from c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62:0,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:142,
                 from C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.h:8,
                 from C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:5:
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3451:5: note: candidate: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3451:5: note:   template argument deduction/substitution failed:
C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:179:65: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
     for (uint8_t i=0; i < min(answer_length, OBD9141_BUFFER_SIZE); i++){
                                                                 ^
In file included from c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:62:0,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:142,
                 from C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.h:8,
                 from C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:5:
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3445:5: note: candidate: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)
     min(initializer_list<_Tp> __l)
     ^
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algo.h:3445:5: note:   template argument deduction/substitution failed:
C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:179:65: note:   mismatched types 'std::initializer_list<_Tp>' and 'unsigned char'
     for (uint8_t i=0; i < min(answer_length, OBD9141_BUFFER_SIZE); i++){
                                                                 ^
In file included from c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:61:0,
                 from C:\Users\USER\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:142,
                 from C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.h:8,
                 from C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:5:
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:243:5: note: candidate: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:243:5: note:   template argument deduction/substitution failed:
C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:179:65: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
     for (uint8_t i=0; i < min(answer_length, OBD9141_BUFFER_SIZE); i++){
                                                                 ^
In file included from c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:61:0,
[... cut ...]
     min(const _Tp& __a, const _Tp& __b)
     ^
c:\users\user\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:195:5: note:   template argument deduction/substitution failed:
C:\Users\USER\Documents\Arduino\libraries\OBD9141-master\src\OBD9141.cpp:179:65: note:   deduced conflicting types for parameter 'const _Tp' ('unsigned char' and 'int')
     for (uint8_t i=0; i < min(answer_length, OBD9141_BUFFER_SIZE); i++){

iwanders, 2022-07-16:
> when i uncomment #define OBD9141_DEBUG i get this errors (sorry for the long list) 

I edited your comment, you can open code blocks with triple backticks on a line (`) and close them again with three backticks, edit your comment to see how I made it more readable.

Only the first part of the error is really important here, the key is `min(uint8_t&, int)`. The `min()` call here in the library is based on the ones arduino supports, which probably also provide overloads for `min(int, int)`, the toolchain you are you using only supports the [templated min](https://en.cppreference.com/w/cpp/algorithm/min) function and because we pass in two different arguments the template argument deduction (so what type the compiler uses for the function) fails because the arguments are different types. You can probably fix this by changing it such that an explicit template argument is provided:
```cpp
for (uint8_t i=0; i < min<uint8_t>(answer_length, OBD9141_BUFFER_SIZE); i++)
```

If that doesn't work, we can also cast the first argument to be similar to the second, the second is a define of an integer, so it defaults to `int`, in that case we would do:
```cpp
for (uint8_t i=0; i < min(static_cast<int>(answer_length), OBD9141_BUFFER_SIZE); i++)
```

Hopefully this unblocks you and allows you to use the debug functionality, hopefully it can help you get things working!

hulktech, 2022-07-18:
Everythink worked perfectly : successfully connected to obd2 and Debuging runing corect;y !! For Debuging is use <uint8_t> for min.
Thank you very much for your help :)

iwanders, 2022-07-18:
@hulktech , I filed  #36 to remove the `min()` function call altogether, I think it is redundant, see the description of the PR. It'd be appreciated if you could test that branch / version of the file just to have a second sanity check besides my own.

(Also, great it's all working now, I'm always happy to hear people successfully use this library ^_^)

hulktech, 2022-07-19:
After i had replaced the line 179 ,the compiler completed successfully ,also checked on a vehicle and work correctly with debug mode of course.
Nice work.

iwanders, 2022-07-19:
I've merged #36, closing this issue as things are reported working and we've improved the library to make it more portable. Thanks @hulktech for reaching out.

closed 2022-07-19 by iwanders