iwanders/OBD9141 #37: problem  'serial_port' was not declared in this scope

gauix4, 2022-09-11:
Hello
impossible to operate there is always an error code
src/main.cpp: In function 'void setup()':
src/main.cpp:47:13: error: 'serial_port' was not declared in this scope
   obd.begin(serial_port, RX_PIN, TX_PIN);

I really don't understand where the error comes from I am using an Arduino Uno at the moment I will switch to an esp32 later
thank you in advance for your answer which will be very valuable for me and for other people because I have already seen the problem pass on several places thank you again

gauix4, 2022-09-11:
[image]

gauix4, 2022-09-13:
`#include <Arduino.h>
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // R,T
#define TX 11

byte ECUaddress = 0x01;

bool Init = 0;

byte Parameter;
byte b;
uint8_t number = 0;
byte counter = 0x01;
byte countPID = 0;
byte countID = 0;

int RPM = 0;

void send5baud(uint8_t data);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("---KWP 5 baud init"));
  delay(500);
  pinMode(TX, OUTPUT);
  digitalWrite(TX, HIGH); // BUS IDLE
  delay(500);

  send5baud(ECUaddress); // функция отправки адреса блоку
  mySerial.begin(9600);
}

void loop()
{

  while (mySerial.available())
  {
    delay(5);

    byte inByte = mySerial.read();
    Serial.print(" Rx: ");
    Serial.print(inByte, HEX);
  }
}

`

gauix4, 2022-09-13:
thank you very much for your answer ! :) 

it's still not able to communicate with my car no sign of life i'm still going with an arduino uno at the moment i haven't tried through software
Here is the code I am using now

`#include <Arduino.h>
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // R,T
#define TX 11

byte ECUaddress = 0x01;

bool Init = 0;

byte Parameter;
byte b;
uint8_t number = 0;
byte counter = 0x01;
byte countPID = 0;
byte countID = 0;

int RPM = 0;

void send5baud(uint8_t data);

void setup()
{
  Serial.begin(115200);
  Serial.println(F("---KWP 5 baud init"));
  delay(500);
  pinMode(TX, OUTPUT);
  digitalWrite(TX, HIGH); // BUS IDLE
  delay(500);

  send5baud(ECUaddress); // функция отправки адреса блоку
  mySerial.begin(9600);
}

void loop()
{

  while (mySerial.available())
  {
    delay(5);

    byte inByte = mySerial.read();
    Serial.print(" Rx: ");
    Serial.print(inByte, HEX);
  }
}

//ниже отправка блоку байта адреса на 5бод  1 стартбит, 7бит данных, 1 бит проверки четности odd, 1 стоп бит

void send5baud(uint8_t data)
{
// // 1 start bit, 7 data bits, 1 parity, 1 stop bit
#define bitcount 10
  byte bits[bitcount];
  byte even = 1;
  byte bit;
  for (int i = 0; i < bitcount; i++)
  {
    bit = 0;
    if (i == 0)
      bit = 0;
    else if (i == 8)
      bit = even; // computes parity bit
    else if (i == 9)
      bit = 1;
    else
    {
      bit = (byte)((data & (1 << (i - 1))) != 0);
      even = even ^ bit;
    }
    Serial.print(F("bit"));
    Serial.print(i);
    Serial.print(F("="));
    Serial.print(bit);
    if (i == 0)
      Serial.print(F(" startbit"));
    else if (i == 8)
      Serial.print(F(" parity"));
    else if (i == 9)
      Serial.print(F(" stopbit"));
    Serial.println();
    bits[i] = bit;
  }
  // now send bit stream
  for (int i = 0; i < bitcount + 1; i++)
  {
    if (i != 0)
    {
      // wait 200 ms (=5 baud), adjusted by latency correction
      delay(200);
      if (i == bitcount)
        break;
    }
    if (bits[i] == 1)
    {
      // high
      digitalWrite(TX, HIGH);
    }
    else
    {
      // low
      digitalWrite(TX, LOW);
    }
  }
  mySerial.flush();
}
`

gauix4, 2022-09-16:
I really thank you very much for answering my questions it's very kind of you ! 

I should rather start my project from the beginning
I will use an esp32 to make a web server that I will use in dashboard
I will also log with the car the car is a Mini Cooper S R53 K-line ISO9141-2
I'm not a great coder I manage but when it comes to tram or timer I'm lost I don't know yet
so I looked a bit to modify it and make it compatible with an esp32 but I can't do it I don't know how to code enough for it
the advantage of esp32 is that there are two serial inputs
this one is pretty fast and has a wifi and bluetooth module that would be perfect for the project and i know very well how to use it to make a web server with it
I'm thinking maybe put an Arduino uno and send the data to the esp32 if I can't do the fusion

sorry if my english is not correct i use google translate lol

iwanders, 2022-09-16:
I haven't heard from anyone who tried to use this library on an esp32. If it has two serial ports I don't see any reason why it wouldn't work. If you do get it to work let us know :)

gauix4, 2022-09-17:
ok :)
what software do you use to parallel read the K-line ?

iwanders, 2022-09-17:
I own a [saleae](https://www.saleae.com/) logic analyzer, one of the first generation Logic 8 devices. You can find some Saleae captures in the [logic](https://github.com/iwanders/OBD9141/tree/master/logic) directory. But the K-line is a slow data bus, so any logic analyzer should work really. Be sure to use a voltage divider to drop lower from 12v to 5v or 3.3v before connecting anything.

gauix4, 2022-09-17:
I'm fine giving up everything because I'm really fed up I don't have the slightest sign of life I started everything from scratch I tried several montages nothing works I changed an Arduino mega to esp32 to uno that still the same no sign of life

gauix4, 2022-09-24:
Here is the working code for an Arduino Uno
suddenly I solved my problem there were two the code and the plate is not compatible with ISO9141

```
#include <Arduino.h>

#include "AltSoftSerial.h"

#include "OBD9141.h"

#define RX_PIN 8  // connect to transceiver Rx
#define TX_PIN 9  // connect to transceiver Tx

AltSoftSerial altSerial;

OBD9141 obd;

bool printSupportedPID(uint32_t value, uint8_t offset);

void setup(){
    Serial.begin(115200);
    obd.begin(altSerial, RX_PIN, TX_PIN);

}

void loop(){
    Serial.println("Looping");

    bool init_success =  obd.init();
    Serial.print("init_success:");
    Serial.println(init_success);

    if (init_success){
        bool res;
        while(1){
            res = obd.getCurrentPID(0x11, 1);
            if (res){
                Serial.print("Result 0x11 (throttle): ");
                Serial.println(obd.readUint8());
            }
            
            res = obd.getCurrentPID(0x0F, 1);
            if (res){
                Serial.print("Intake air temperature ");
                Serial.println(obd.readUint8());
            }

            res = obd.getCurrentPID(0x0B, 1);
            if (res){
                Serial.print("absolute pressure ");
                Serial.println(obd.readUint8());
            }

            res = obd.getCurrentPID(0x05, 1);
            if (res){
                Serial.print("Engine coolant temperature ");
                Serial.println(obd.readUint8());
            }

            res = obd.getCurrentPID(0x42, 1);
            if (res){
                Serial.print("volte ");
                Serial.println(obd.readUint8());
            }
            Serial.println();

            delay(200);
        }
        delay(200);
    }
    delay(3000);

} 
```

gauix4, 2022-09-24:
So I switched to an esp32 and it works fine.
on the other hand the refresh rate is very slow about every second I have not found or improved this I want to make a logger so it takes much faster data at least every 100ms
thank you very much for answering me so far it's very kind of you and thank you again for this library

iwanders, 2022-09-24:
:tada: , glad you got it working. Looks very similar to [reader_softserial](https://github.com/iwanders/OBD9141/blob/master/examples/reader_softserial/reader_softserial.ino) in the end :)

> on the other hand the refresh rate is very slow about every second I have not found or improved this I want to make a logger so it takes much faster data at least every 100ms

Unfortunately, there's not much that can be done about that, all timing depends on the ECU. See the [timing section](https://github.com/iwanders/OBD9141#timing) of the readme, everything in the library is as fast as it can be and we never delay when reading. In all places `readBytes` is used which returns as soon as the desired number of bytes has been obtained.

gauix4, 2022-09-25:
strange that it does not go faster because I have a special k-line adapter for my mini I log with it and it is very fast
on the mini there is no Can bus on the OBD socket but only k-line
we can see on the diagram there is only the case line or the DS2

Here is the adapter I am using
https://www.amazon.com/Diagnostic-Interface-FT232RL-Coding-Programing/dp/B08C2WW33G

[image]

iwanders, 2022-09-25:
> Here is the adapter I am using
> https://www.amazon.com/Diagnostic-Interface-FT232RL-Coding-Programing/dp/B08C2WW33G

I wonder if that can speak K-line with a different baudrate; `Slow magistral K-CAN (100 kbit/s)` that's 100000 bit/s, which exceeds the [10400](https://github.com/iwanders/OBD9141/blob/6b4a73ac9751077395611c581c391b6bf198667a/src/OBD9141.h#L22) used and specified in the standard.

I've never heard of a K-line with a higher bus rate. Without data recordings / logic analyzer / further information it'll be hard to figure out how this device gets such high speeds. Perhaps you can make a logic analyzer from an [esp32](https://github.com/EUA/ESP32_LogicAnalyzer)? Use the adapter from amazon, and connect the logic analyzer to the Rx signal.

gauix4, 2022-09-26:
more here #40 :)