iwanders/OBD9141 #38: For ESP32

gauix4, 2022-09-24:
```cpp
#include <Arduino.h>
#include "OBD9141.h"

#define RX_PIN 16 // 16  // connect to transceiver Rx for ESP32
#define TX_PIN 17  // 17  // connect to transceiver Tx

bool init_success;

OBD9141 obd;

bool printSupportedPID(uint32_t value, uint8_t offset);

void setup()
{

  Serial.begin(115200);
  obd.begin(Serial2, RX_PIN, TX_PIN);
  delay(2000);
  Serial.println("initialization");
  Serial.println("");
  
  while (!obd.init())
  {
     return;
  }
}

void loop()
{

  if (obd.getCurrentPID(0x11, 1))
  {
    Serial.print("Result 0x11 (throttle): ");
    Serial.println(obd.readUint8());
  }

  if (obd.getCurrentPID(0x0F, 1))
  {
    Serial.print("Intake air temperature ");
    Serial.println((obd.readUint8()) / 5);
  }

  if (obd.getCurrentPID(0x0B, 1))
  {
    Serial.print("absolute pressure ");
    Serial.println(obd.readUint8());
  }

  if (obd.getCurrentPID(0x05, 1))
  {
    Serial.print("Engine coolant temperature ");
    Serial.println((obd.readUint8()) / 5);
  }

  if (obd.getCurrentPID(0x42, 1))
  {
    Serial.print("volte ");
    Serial.println(obd.readUint8());
  }
  
  Serial.println("");
}
```

closed 2022-09-24 by gauix4

iwanders, 2022-09-24:
Awesome, I made PR for it; https://github.com/iwanders/OBD9141/pull/39

Could you comment there whether that branch works for you?

gauix4, 2022-09-25:
this one works great i am using it right now
I didn't know how you suggested it
so i posted it here sorry

iwanders, 2022-09-25:
> I didn't know how you suggested it

All good. If you push a new branch to your fork of the repo, you could file a pull request on github. But this worked as well, thanks for sharing this example.