iwanders/OBD9141 #44: How can I know if I'm doing everything right?

tzomats, 2022-11-01:
Hi, I'm new to this topic and would like some clearer guidance. what I mean!
if there is any how to video or pictures how to make the connection between Teensy 3 and the MC33290, SN65HVDA100 and SN65HVDA195 and the plug of my car?
my car is fiat panda 2008.
I know I'm asking a lot but any help would be a lifesaver for me

iwanders, 2022-11-01:
> if there is any how to video or pictures how to make the connection between Teensy 3 and the MC33290, SN65HVDA100 and SN65HVDA195 and the plug of my car?

The [usage](https://github.com/iwanders/OBD9141#usage) section has a diagram on how to connect the SN65HVDA195 to your Teensy. Use the Teensy's TX pin 1, and RX pin 0, connect enable to pin 0 as per the [reader](https://github.com/iwanders/OBD9141/blob/master/examples/reader/reader.ino) example. 

As for the plug of your car, [wikipedia](https://en.wikipedia.org/wiki/On-board_diagnostics#OBD-II_diagnostic_connector) has a diagram on which pin is what. You should only need the K-line of ISO 9141-2 and the battery voltage. Be sure to wire it up like the diagram from the readme, with the resistor and diodes. At rest, the Kline should be at battery voltage.

After you're confident in your physical connections, read #3  for how to enable debug output, such that you get some more insight into what's happening. If you aren't 100% sure what handshake your car uses, also try [readerKWP](https://github.com/iwanders/OBD9141/blob/master/examples/readerKWP/readerKWP.ino) and [readerKWPSlow](https://github.com/iwanders/OBD9141/tree/master/examples/readerKWPSlow).

tzomats, 2022-11-02:
> The [usage](https://github.com/iwanders/OBD9141#usage) section has a diagram on how to connect the SN65HVDA195 to your Teensy. Use the Teensy's TX pin 1, and RX pin 0, connect enable to pin 0 as per the [reader](https://github.com/iwanders/OBD9141/blob/master/examples/reader/reader.ino) example.

first of all thank you for answering me!
for me, who don't know it, I find it difficult to understand the circuit.
I will start the process if I need anything I hope to have the valuable help from you

tzomats, 2022-11-02:
is poseble to work with arduino nano, that I have at the moment?

tzomats, 2022-11-02:
> If you find this circuit difficult to understand, this project may be a bit outside of your skill level. In general things don't work 'out of the box',

don't worry about that, no one was born to know!
I also have a friend who can help me on the circuit!
thank you very much