iwanders/OBD9141 #45: esp32 with OBD9141 not work

taoloo897, 2023-02-16:
HI IWANDERS.
I tried from 2 days to make it works but without succes!
I use esp32 wroomer 32U and a mc33290. Witrh this setup i had other library witch work at the same pinrx 16 and tx 17. For mc33290 i used a resistor 500k between 12v and k-line. I work with ECU me7.5.

If i use reader esp32  with INIT i receive :
16:17:43.716 -> Before magic 5 baud.
16:17:45.697 -> Before setting port.
16:17:45.732 -> After setting port.
16:17:45.732 -> First read is: 252
16:17:45.797 -> Timeout on reading bytes.
16:17:45.929 -> Timeout on reading bytes.
16:17:46.027 -> Timeout on reading bytes.
16:17:46.027 -> 

If i tried to use INITKWPSlow
16:29:55.011 -> initialization
16:29:55.011 -> 
16:29:58.026 -> Before 25 ms / 25 ms startup.
16:29:58.091 -> Enable port.
16:29:58.418 -> 
16:29:58.646 -> 

Please waiting your advices.
Many thanks.

iwanders, 2023-02-16:
> With this setup i had other library witch work at the same pinrx 16 and tx 17

Do you know which protocol was used by that library? What library was it? If you have something that works, why switch? :) 

Did you also try the [readerKWP](https://github.com/iwanders/OBD9141/blob/master/examples/readerKWP/readerKWP.ino)? Fast kwp init and slow kwp init are different, I don't know what you ECU uses to communicate, but you showed 9141-2 init with the slow 5 baud and initKWPSlow not working, might as well try the third and final handshake this library supports.

You can also enable the debug prints by uncommenting this [line](
https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.h#L11).

taoloo897, 2023-02-17:
Hi.
I had 2 libraries one with kwp2000 and other with 9141-2 , both works very slow and intrerrupt my lcd display data. Both use 5-baud.
I tried readKWP , i uncomment ""debug. , you can see upstairs what i capture on serial console.
With 9141-2 i receive 16:17:45.732 -> First read is: 252 , not read from ecu 0x55.
Thanks in advance

iwanders, 2023-02-17:
> I had 2 libraries one with kwp2000 and other with 9141-2 , both works very slow and intrerrupt my lcd display data. Both use 5-baud.

This library is also synchronous though, so it will also block your main program until the data comes back from the ECU. You will likely have the same experience.

> With 9141-2 i receive 16:17:45.732 -> First read is: 252 , not read from ecu 0x55.

Ah, 252 is `0xfc`, that is not the `0x55` that is [expected](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.cpp#L386) and should be the first byte sent after the handshake. You can try commenting out line 386, and seeing if the other two checks on line 402 and 411 pass.

You could also put print statements in the libraries that are working, to see how their handhake works.

taoloo897, 2023-02-17:
Thanks for your time.
I comment the line 386 and nothing happen
15:43:21.733 -> initialization
15:43:21.734 -> 
15:43:23.733 -> Before magic 5 baud.
15:43:25.736 -> Before setting port.
15:43:25.736 -> After setting port.
15:43:25.736 -> First read is: 252
15:43:25.769 -> Timeout on read v1.
15:42:31.796 -> Timeout on reading bytes.
15:42:31.893 -> Timeout on reading bytes.
15:42:31.992 -> Timeout on reading bytes.
15:42:31.992 -> 
15:42:32.090 -> Timeout on reading bytes.
15:42:32.186 -> Timeout on reading bytes.
15:42:32.283 -> Timeout on reading bytes.
15:42:32.283 ->

iwanders, 2023-02-17:
> Must this library work with mc33290 too , right?

Certainly, I've used that chip myself with this library.

> I comment the line 386 and nothing happen
Looks like just nothing comes back. Tricky! I'd try to record a working handshake with one of the other libraries with a logic analyser, or even just print statements, that way there's something to compare against.

taoloo897, 2023-02-18:
Hi again. this is what one of library had on serial/debug

07:40:39.979 -> Starting Init
07:40:42.044 -> SYNC Byte: 55
07:40:42.044 -> KEY1: 8    KEY2: 8
07:40:42.076 -> Sending ~KEY2: F7
07:40:42.076 -> Received ~KEY2: F7
07:40:42.108 -> Inverted Address: CC
07:40:42.108 -> Protocol: ISO 9141-2
07:40:42.108 -> 5-baud Init SUCCESS

and sync code is aproximatly the same with your obd9141.cpp 

pinMode( txPin, OUTPUT );   // Set the txPin to idle HIGH

  delay( 2000 ); // Leave Tx HIGH for 2 seconds before starting the actual init routine
 
  Serial.println( F( "Starting Init" ) );
 
  // Send 0x33 (LSB first) via K-Line to the OBD port (0x33 is address of OBD)
  digitalWrite( txPin, LOW ); // start bit
  delay( 200 );               // 5 baud means 200ms per bit
  digitalWrite( txPin, HIGH ); //11  
  delay( 400 );
  digitalWrite( txPin, LOW ); //00  
  delay( 400 );
  digitalWrite( txPin, HIGH ); //11
  delay( 400 );
  digitalWrite( txPin, LOW ); //00
  delay( 400 );
  digitalWrite( txPin, HIGH ); // stop bit  
  delay( 200 );
 mc33290.begin(10400, SERIAL_8N1 , 16 ,17) ; //mc33290 is a hardware serial defined
 if( mc33290.available()){
    sync =mc33290.read();
    //DEBUG
    snprintf_P( buffer, BUFLEN, PSTR("SYNC Byte: %2X"), sync );
    Serial.println( buffer );
 
    // Continue Init if we got the correct SYNC Byte
    if( sync == 0x55 ) {

iwanders, 2023-02-18:
Thanks for sharing code and prints. I put some backticks (`) in your comment to create code blocks to make it more readable.

> OBD9141println(serial_port);

I don't think that print the serial ports' id, I think instead it will try to read a byte from the serial port and print that.

But this one:
```
mc33290.begin(10400, SERIAL_8N1 , 16 ,17) 
```
that looks like a difference from my library, I don't pass the pin numbers, but it looks like they vary between `esp32` and `esp32s3`, according to [this link](https://github.com/espressif/arduino-esp32/blob/2333df5ac650fd9ebc0a1a0d7df93851c86a95e1/cores/esp32/HardwareSerial.cpp#L77-L92), so it could be that you have an `esp32s3` style board and are actually opening the serial port on pin 19 and 20 instead of the 16 and 17 you are expecting.

We can certainly pass these pin numbers, I put up a change with 1d07f97 on the [issue-45-esp32-serial-port](https://github.com/iwanders/OBD9141/tree/issue-45-esp32-serial-port) branch, could you give that a go? I obviously can't test this, but if you are indeed on an esp32s3 board, this could fix it.

taoloo897, 2023-02-20:
Hi and excuse my late answer. I'm not at home and i can verify only after one week because i'm in hollyday. My esp32 is a wroomer 32 u with 32 pinds  i't must doesn't matter witch pins used tge serial port because i configured in sketch at the begin line on other pins.and it mjst work. The problem is from other parts, and we will see. Thanks and keep in touch.

taoloo897, 2023-02-26:
Hi. I arrived at home and i tested the new version of your library.
Doesnt't work the same error:
16:30:05.200 -> initialization
16:30:05.200 -> 
16:30:05.559 -> Before magic 5 baud.
16:30:07.357 -> Before setting port.
16:30:07.357 -> After setting port.
16:30:07.357 -> First read is: 0
16:30:07.389 -> Timeout on read v1.
16:30:07.485 -> Timeout on reading bytes.
16:30:07.584 -> Timeout on reading bytes.

How can we see if the serial2 is correct init?

iwanders, 2023-02-26:
> How can we see if the serial2 is correct init?

Not sure if there's any way... :/ 

Start by putting a [print here](https://github.com/iwanders/OBD9141/blob/1d07f979494d19714d1e0d9bb96d48c1ce4adb8f/src/OBD9141.cpp#L31) to print the pin numbers that are being set, that also confirms the `ESP_ARDUINO_VERSION_MAJOR` define is actually set and we are going into that initialisation.

taoloo897, 2023-02-26:
YES, enter in that ESP_ARDUINO_VERSION_MAJOR

iwanders, 2023-02-26:
Hmm, well, the ECU is clearly not talking back. The first read isn't even `252` anymore like it was before. The hardware is the exact same as in [this comment](https://github.com/iwanders/OBD9141/issues/45#issuecomment-1435493682)?  Because this is really weird.

If you have a second arduino & transceiver chip, you can try to listen in on the communication performed on the K-line by just hooking up the Rx and opening a serial port at 10400 baud, that way you should be able to see what the other library sends on the bus and what this one sends. It also allows ruling out eletrical issues.

taoloo897, 2023-02-26:
Yes it is exactly the same. If i upliad thst library over yours in my esp32 it works..
I will try with an arduino megs these days , first with your library, second as listen, but i don't know exactly how. Between another arduino what transceiver or how make connection. Thanks again.

taoloo897, 2023-03-12:
HI again. I made teste and tests..... reader_esp32 not work, i receive first byte every time 0.
I tried readerKWPSlow because i have like (1941) other file witch works.
I modified delay times in InitTmplm but not works, received first byte 0.
bool OBD9141::initImpl(bool check_v1_v2){
    use_kwp_ = false;
    // this function performs the ISO9141 5-baud 'slow' init.
    this->set_port(false); // disable the port.
    this->kline(true);
	delay(350);
   // delay(OBD9141_INIT_IDLE_BUS_BEFORE); // no traffic on bus for 3 seconds.
    OBD9141println("Before magic 5 baud.");
    // next, send the startup 5 baud init..
    this->kline(false); delay(200); // start
    this->kline(true); delay(200);  // first two bits
    this->kline(false); delay(600); // second pair
    this->kline(true); delay(200);  // third pair
    this->kline(false); delay(600); // last pair
    //this->kline(true); delay(200);  // stop bit
    // this last 200 ms delay could also be put in the setTimeout below.
    // But the spec says we have a stop bit.

STRANGE what is next:
I put only some degub in begin function and set_port like this  (marked with //added) , i think it they make some delay:
void OBD9141::begin(OBD_SERIAL_DATA_TYPE & serial_port, uint8_t rx_pin, uint8_t tx_pin){
    this->serial = &serial_port;
    this->tx_pin = tx_pin;
    this->rx_pin = rx_pin;

    OBD9141println("1");//added

    // Enable the pullup on the Rx Pin, this is not changed by set_port.
    pinMode(this->rx_pin, INPUT);
    digitalWrite(this->rx_pin, HIGH);
    this->set_port(true); // prevents calling this->serial->end() before start.
    use_kwp_ = false;

    OBD9141println("2");//added
}

void OBD9141::set_port(bool enabled){
    if (enabled){
  
	 OBD9141println("3"); //added

        // Work around the incorrect pinmode configuration in Due.
        #ifdef ARDUINO_SAM_DUE
          g_APinDescription[this->rx_pin].pPort -> PIO_PDR = g_APinDescription[this->rx_pin].ulPin;
          g_APinDescription[this->tx_pin].pPort -> PIO_PDR = g_APinDescription[this->tx_pin].ulPin;
        #endif
        this->serial->begin(OBD9141_KLINE_BAUD);
    } else {
        this->serial->end();
	    OBD9141println("4");//added
        #ifdef ARDUINO_SAM_DUE
          g_APinDescription[this->rx_pin].pPort -> PIO_PER = g_APinDescription[this->rx_pin].ulPin; 
          g_APinDescription[this->tx_pin].pPort -> PIO_PER = g_APinDescription[this->tx_pin].ulPin; 
        #endif
        pinMode(this->tx_pin, OUTPUT);
        digitalWrite(this->tx_pin, HIGH);
    }
}

AND , AND i received this.....what is ok for KWP

17:16:19.124 -> Before magic 5 baud.
17:16:20.932 -> Before setting port.
17:16:20.932 -> 3
17:16:20.932 -> After setting port.
17:16:21.193 -> First read is: 85
17:16:21.225 -> read v1: 239
17:16:21.225 -> read v2: 143
17:16:21.225 -> v1: 239
17:16:21.225 -> v2: 143
17:16:21.290 -> init_success:0
Init return false because remaining last part of initTmpl  where after this next line nothing happen, i put a debug again and....not go from this line

// these two should be identical according to the spec.
    OBD9141println("CCCC0"); //added  - HERE STOPED AND first byte is 0. STRANGEGEGEGEGEGEGEGEGEGEGEGEGEG
    
//here never arrived
    if (check_v1_v2) {
      if (v1 != v2){
          return false;
      }
    }

17:26:20.798 -> Before magic 5 baud.
17:26:22.603 -> Before setting port.
17:26:22.603 -> 3
17:26:22.603 -> After setting port.
17:26:22.603 -> First read is: 0
17:26:22.603 -> init_success:0

iwanders, 2023-03-12:
Hmm, looks like you're on the right track, things are coming back now.

In `initKWPSlow` we actually disable the `v1 == v2` check [here](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.cpp#L458), since you get a `v2` of 143, and that corresponds to the comment there, I think things are working.

Try just disabling [this check](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.cpp#L421) by setting `check_v1_v2 = false`.

taoloo897, 2023-03-13:
it is setted to false, not enter in that check, after that check is a problem, if i put a debug (print) funtion ....the program stopped

iwanders, 2023-03-13:
Well, if the program suddenly stops if you put a print statement somewhere... I think there are other problems in your program / setup. Not much I can do to help with that though.

taoloo897, 2023-03-13:
It is not from the sketch, it is something inside the library.. If you see first post from yesterday, something is strange. These OBD9141println("")  added in OBD9141.begin and OBD9141.set_port add some delay and like i said with them added in the code i can receive good response with KWPSlow.

iwanders, 2023-03-13:
> OBD9141println("") added in OBD9141.begin and OBD9141.set_port add some delay

What happens if you replace that with a `delay(10)` or something? It sounds like a timing issue to me, not like something is fundamentally wrong with the library.

iwanders, 2025-04-05:
This is a bit late, but #63  likely resolves this, debugging happened in #62. I'm going to close this as I expect it's now resolved.

closed 2025-04-05 by iwanders