iwanders/OBD9141 #46: A special kind of communication with the ECU

Vladislav-Lavrushkin, 2023-04-19:
Hello, My control unit has special modes and PID, how can I work with this with your library?
[image]
This is a request from the CASCADE 0.9.4 program.

Vladislav-Lavrushkin, 2023-04-19:
I make such a request but get no response.
[image]

Vladislav-Lavrushkin, 2023-04-19:
For information, 2 requests. The one on the right is understandable. The one on the left is not clear.
[image]

Vladislav-Lavrushkin, 2023-04-20:
I begin to understand.
The request does not consist of 5 bytes, but of 6 bytes.

Vladislav-Lavrushkin, 2023-04-20:
Now you need to understand how to send a request of 6 bytes:)

Vladislav-Lavrushkin, 2023-04-20:
`bool OBD9141::getlongPID(uint8_t pid, uint8_t mode, uint8_t return_length){
    uint8_t message[6] = {0x68, 0x6A, 0xF1,0x22, mode, pid};
    // header of request is 5 long, first three are always constant.

    bool res = this->request(&message, 6, return_length+6);
    // checksum is already checked, verify the PID.

    if (this->buffer[5] != pid){
        return false;
    }
    return res;
}`

I wrote a function but it doesn't work. Help me please.

Vladislav-Lavrushkin, 2023-04-20:
```
bool OBD9141::getlongPID(uint8_t pid, uint8_t mode, uint8_t return_length){
    uint8_t message[6] = {0x68, 0x6A, 0xF1,0x22, mode, pid};
    // заголовок 6 байт

    bool res = this->request(&message, 6, return_length+6);
    // checksum is already checked, verify the PID.

    if (this->buffer[5] != pid){
        return false;
    }
    return res;
}
```
I have added a function to the library.
But I get "60 00" in response.
How can I shift the data to the right by 1 byte? I see part of the request.

iwanders, 2023-04-25:
I was away from my computer for a few weeks, glad you got it to work by yourself, looks great!

The [request](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.h#L151) method is public, so you could also use a helper function outside of the object, if you want to use the library unmodified. You can retrieve the buffer with [readBuffer](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.h#L184), which is also public. This is the first time I've seen a longer request, so I don't think there's much value in adding it to the library.

@Vladislav-Lavrushkin , is this issue good to be closed?

Vladislav-Lavrushkin, 2023-04-25:
Yes, the topic can be closed.

closed 2023-04-25 by iwanders