iwanders/OBD9141 #47: ESP32 Serial1 crash...

janokylaszlo, 2023-05-23:
Dear Iwanders,

I would like to use OBD9141 on an ESP32 SIM800C board, where the 16/17 pins are not available. Unfortunately the Serial1 RX_PIN 4 and TX_PIN 2 configurations crash :( What am I doing wrong?

Thanks a lot!
LJ

The code:
```
#define RX_PIN 4 // connect to transceiver Rx for ESP32
#define TX_PIN 2 // connect to transceiver Tx

bool init_success;
OBD9141 obd;

void setup(){
    Serial.begin(115200);
    obd.begin(Serial1, RX_PIN, TX_PIN);
```

Serial output:
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
ets Jul 29 2019 12:21:4

janokylaszlo, 2023-05-23:
I found the reason: 
the serial initialization is done in line 29. of OBD9141.cpp, but here rx_pin and tx_pin are not included in the this->serial->begin() call, so the default pins are assigned. This code: 
`this->serial->begin(OBD9141_KLINE_BAUD, this->tx_pin, this->rx_pin);`
already sets the pins assigned to the serial.

iwanders, 2023-05-24:
Yeah, so the object only needs tx and rx to be able to set the pullups. The problem is that the arduino ecosystem only specifies the baudrate as [argument](https://reference.arduino.cc/reference/tr/language/functions/communication/serial/begin/) for `begin()`, the pin numbers are usually implicit and based on whether `Serial`, `Serial1` or `SerialN` is used.

This is also why the [esp example](https://github.com/iwanders/OBD9141/blob/master/examples/reader_esp32/reader_esp32.ino) doesn't set the pins either. The esp api deviates a bit here in supporting it I think...

Even setting them once with a `begin()` call outside of the `set_port` function wouldn't work. [they get reset to defaults](https://github.com/espressif/arduino-esp32/blob/3ec5f4efa1de4342aaab742008dc630091e5e035/cores/esp32/HardwareSerial.cpp#L343) if you don't pass them when calling `begin()` after an `end()` call.

I guess we could wrap it in a preprocessor directive, I'd be happy to accept a PR that updates this and only for ESP32s performs `this->serial->begin(OBD9141_KLINE_BAUD, this->tx_pin, this->rx_pin);` instead of the `begin()` call without arguments.

iwanders, 2025-04-05:
This is a bit late, but #63  likely resolves this, debugging happened in #62. I'm going to close this as I expect it's now resolved.

closed 2025-04-05 by iwanders