iwanders/OBD9141 #48: KWP71 Compatibility

dimiras, 2023-07-12:
Do you think this would work with an ECU that uses the KWP71 used on some older bmw ecus? 

I think its based on the ISO9141.

Here is an image with some info about the protocol:
[image]

And also the page where I found this information:
http://www.km5tz.com/BMW%20850iP11.htm

I also found this repository that also mentions the ISO9141:
https://github.com/kaihara/kwp71scan/wiki

Also is there any board that I can use directly with arduino? without the need of soldering?

dimiras, 2023-07-12:
Also found this http://www.internetsomething.com/kwp/

iwanders, 2023-07-12:
> I think its based on the ISO9141.
> I also found this repository that also mentions the ISO9141:

It mentions ISO9141 because the physical bus characteristics are similar / identical. The protocol itself looks different; 9600 as baudrate, every byte is echo'd back inverted and it doesn't look like there is a checksum, all of which are key differences to ODB9141-2.

> Also is there any board that I can use directly with arduino? without the need of soldering?

The `SN65HVDA` series chips are usually K-line transceiver chips, so that could very well work if the physical bus is identical.

> Do you think this would work with an ECU that uses the KWP71 used on some older bmw ecus?

I don't think this library will really work for this protocol. I'd start from scratch, or use the [kwp71scan](https://github.com/kaihara/kwp71scan) repo you found, the actual protocol implementation seems to be [grouped in these free functions](https://github.com/kaihara/kwp71scan/blob/dcae15a8c089af3108406a04a9fd2ed06e4af776/scan/scan.ino#L172-L378), so you may be able to pull that out without too much problems and reuse it.

Good luck with your project!

closed 2023-07-12 by iwanders