iwanders/OBD9141 #49: Unable to initialize obd.init()

Ash-S, 2023-08-16:
Hi @iwanders Thanks for your sharing. https://github.com/iwanders/OBD9141
I am trying example code 'reader_softserial' but unable to init the obd as defined here in obd.init().
Please suggest~ Thanks.

17:54:21.677 -> init_success:0
17:54:25.191 -> Looping
17:54:31.178 -> init_success:0
17:54:34.677 -> Looping
17:54:40.687 -> init_success:0
17:54:44.166 -> Looping
17:54:50.169 -> init_success:0
17:54:53.671 -> Looping

iwanders, 2023-08-16:
Start by toggling on the debug printing [here](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.h#L11).

If possible and you have the hardware, attach a logic analyzer, compare with another OBD reader.

Ash-S, 2023-08-17:
Before magic 5 baud.
 Before setting port.
After setting port.
 Timeout on read 0x55.
 init_success:0
 Looping
 Before magic 5 baud.
 Before setting port.
 After setting port.
 Timeout on read 0x55.
 init_success:0

iwanders, 2023-08-17:
No, you're getting a timeout; [this line](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.cpp#L383) returns false.

Are you sure this is the correct initialisation procedure for this ECU? Does it need the 5 baud init or KWP? Ideally confirm with another OBD reader what initialisation is necessary for this ECU.

Check if your software serial port actually works, can you receive anything on this serial port? Does it need pullup resistors?

Ash-S, 2023-08-18:
Yes, I agree, [this line] returns false. basically my ECU is a simulator and I can set the modes using switches to KWP and ISO9141-2.  So I am trying with init() and initKWPSlow(). both needs the 5 baud init. I tested it with another obd reader it detect the KWP and ISO9141-2. Also I compared the wakeup signal from obd reader and what I am getting. Its similar, but after that it doesn't show any data signals (like 1st byte which should be 0x55).

iwanders, 2023-08-19:
Hmm, well that's an intriguing setup.

I wonder what the timing looks like with another OBD reader. A wild guess is that the simulator responds very quickly, effectively responding with the `0x55` in between [these calls](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.cpp#L368-L383), so perhaps the `0x55` is sent while we are still in the `delay(200)` and the serial port is not yet enabled?

You could try to remove that last delay, it shouldn't affect the initialisation procedure since the line is already high, but it may just setup the bus fast enough to be able to catch the `0x55`?

If at all possible, hook up a logic analyser and record a handshake from another OBD reader and compare it with what the handshake from your microcontroller & this library looks like, that is by far the easiest way to diagnose any issues around it.