iwanders/OBD9141 #50: The example for esp32 does not work.

Vladislav-Lavrushkin, 2023-11-06:
The example for esp32 did not work for me.
I remade the example for Arduino, everything works.
#39

iwanders, 2023-11-06:
I think it depends on which ESP board one uses?

Main difference seems to be the serial setup

https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/examples/reader/reader.ino#L19

and

https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/examples/reader_esp32/reader_esp32.ino#L15

One uses `Serial2`, the other `Serial1`?

Feel free to file a PR to modify the current esp32 example if you feel it can be improved. May also be worth stating in a comment in the example which board you used, and that the serial setup may be different depending on the board used?

Vladislav-Lavrushkin, 2023-11-17:
Okay, I'll try to do this. One day)

Today I tried creating tasks to execute code. Since the ESP32 has 2 cores, tasks can be divided.
One core communicates with the car's ECU. And the second core does the calculations.
If this may be useful to someone, I will attach the code below.
P.S. I'm new to programming and this code is a starting point for the project)
`

#include "OBD9141.h"
OBD9141 obd;

#define RX_PIN 16  // connect to transceiver Rx for ESP32
#define TX_PIN 17  // connect to transceiver Tx
bool init_success;

TaskHandle_t Task1;
TaskHandle_t Task2;

float time_inj = 0;
float rpm = 0;
float q_l_sec = 0;
float l_h = 0;
float l_100 = 0;
float speed_car = 0;

void setup() {
  Serial.begin(115200);
  //ESP_BT.begin(device_name);
  obd.begin(Serial2, RX_PIN, TX_PIN);
  delay(2000);
  Serial.println("initialization");

  //создаем задачу, которая будет выполняться на ядре 0 с максимальным приоритетом (1)
  xTaskCreatePinnedToCore(
    Task1code, /* Функция задачи. */
    "Task1",   /* Ее имя. */
    10000,     /* Размер стека функции */
    NULL,      /* Параметры */
    1,         /* Приоритет */
    &Task1,    /* Дескриптор задачи для отслеживания */
    0);        /* Указываем пин для данного ядра */
  delay(500);

  //Создаем задачу, которая будет выполняться на ядре 1 с наивысшим приоритетом (1)
  xTaskCreatePinnedToCore(
    Task2code, /* Функция задачи. */
    "Task2",   /* Имя задачи. */
    10000,     /* Размер стека */
    NULL,      /* Параметры задачи */
    1,         /* Приоритет */
    &Task2,    /* Дескриптор задачи для отслеживания */
    1);        /* Указываем пин для этой задачи */
  delay(500);

  while (!obd.init()) {
    return;
  }
}

void Task1code(void* pvParameters) {
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for (;;) {

    Serial.println("Looping");

    bool init_success = obd.init();
    Serial.print("init_success:");
    Serial.println(init_success);

    if (init_success) {
      bool res;
      while (1) {
        res = obd.getCurrentPID(0x0C, 2);
        if (res) {
          Serial.print("Result 0x0C (RPM): ");
          Serial.println(obd.readUint16() / 4);
          rpm = obd.readUint16() / 4;
        }
        res = obd.getlongPID(0x60, 0x12, 2);
        if (res) {
          Serial.print("Result (time inj #1): ");
          Serial.println((obd.readUint_long16() * 2));
          time_inj = (obd.readUint_long16() * 2);
        }

        res = obd.getCurrentPID(0x0D, 1);
        if (res) {
          Serial.print("Result 0x0D (speed): ");
          speed_car = (obd.readUint8());
          Serial.println(obd.readUint8());
        }
      }
    }
  }
}

void Task2code(void* pvParameters) {
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for (;;) {
    q_l_sec = (rpm / 60.0) * 2.0 * (time_inj / 1000000.0) * (0.13 / 60.0);
    //полная формула
    // q_l_sec = (rpm / 60.0) * (кол-во цилиндров /  2.0) * (time_inj / 1000000.0) * ((130.0 / 1000.0) / 60.0)

    l_h = q_l_sec * 3600;  //литры в час мгновенное

    l_100 = (l_h / speed_car) * 100.0;  // потуги литры на 100 км тоже потуги

    Serial.print("L/h ");
    Serial.println(l_h);
    Serial.print("L/100 ");
    Serial.println(l_100);
  }
}

void loop() {
}`

Vladislav-Lavrushkin, 2023-11-17:
> I think it depends on which ESP board one uses?

ESP-WROOM-32
[image]

closed 2024-01-23 by Vladislav-Lavrushkin