iwanders/OBD9141 #51: PIDS

ClusterModzCo, 2023-12-12:
Hey there ive been trying to get the pid call to work with no luck. im currectly connected but when calling ny other address then the 3that are in the example i get nothing displaying. like nothing at all the whole line is blan. is there a trick im missing or do i need to some how add to the library in order to read the pid i request? if so mind hinting off how to do so please

iwanders, 2023-12-12:
Not all ECU's support all PID's, you can query your ECU to determine what PIDs it supports. See the [supportedPID](https://github.com/iwanders/OBD9141/blob/master/examples/supported_pids/supported_pids.ino) example.

ClusterModzCo, 2023-12-12:
ok i got the supported pid example up and running but yet im still having trouble understanding. i was able to connect initally but right after is spit out data all it wants to do is loop now and no success? also the data display is 0x1 0x2 0x3 so on follows by a 1 or a 0. im assuming this states whats avalible? im working on a 04 allroad me7.1.1 if that helps

ClusterModzCo, 2023-12-12:
> ok i got the supported pid example up and running but yet im still having trouble understanding. i was able to connect initally but right after is spit out data all it wants to do is loop now and no success? also the data display is 0x1 0x2 0x3 so on follows by a 1 or a 0. im assuming this states whats avalible? after the first connection it loops and will not connect agian unless i turn off the ecu and turn it back on. im working on a 04 allroad me7.1.1 if that helps

init_success:1
PID 0x0  -> 0xBF9FB110
0x1: 1
0x2: 0
0x3: 1
0x4: 1
0x5: 1
0x6: 1
0x7: 1
0x8: 1
0x9: 1
0xA: 0
0xB: 0
0xC: 1
0xD: 1
0xE: 1
0xF: 1
0x10: 1
0x11: 1
0x12: 0
0x13: 1
0x14: 1
0x15: 0
0x16: 0
0x17: 0
0x18: 1
0x19: 0
0x1A: 0
0x1B: 0
0x1C: 1
0x1D: 0
0x1E: 0
0x1F: 0
0x20: 0
Looping
init_success:0

iwanders, 2023-12-12:
> also the data display is 0x1 0x2 0x3 so on follows by a 1 or a 0. im assuming this states whats avalible? 

Yep, the 0 denotes not available, the 1 denotes available. This [section on PID 01](https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_00_-_Show_PIDs_supported) on wikipedia is most helpful, it describes how to interpret the response to PID 00 (the 
`0xBF9FB110` you got).

So now you can look in the [show current data](https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01) section of wikipedia to see what data you can retrieve, and how to interpret it.  For example, your car reports `0x10: 1`, so `PID 0x10` (first column in the table) shows that you can retrieve the mass air flow rate. The other columns describe how to interpret that data.

ClusterModzCo, 2023-12-12:
Ok so I do understand all of that I guess the biggest question now at the
moment is why can I not see anything past 20 let's say such as 0x42 control
module voltage or other pids?

iwanders, 2023-12-13:
That `show current data` section on wikipedia is basically the PID description table, if you look there of `0x20`, you'll see that PID `0x20` has the description `PIDs supported [$21 - $40]`, given that PID `0x00` states that `0x20` is not supported, there's no PID's available on your ECU beyond the first block of `0x00-0x20`.

ClusterModzCo, 2023-12-13:
When calling display.println(getCurrentPID(0x10,  2)); (address, byte
length) no matter available or not the return value displayed is 1 is there
a way to call the actual PID that I'm not understanding?

Also on a nother ECU Im trying to connect to is the med9.1 2007 but I'm
unable to initialize a connection with that ECU though it is canbus if I
remember correctly the Kline is the way to read dtc and clear them. I know
that parts a bit off subject but yet I will need to figure out the Kline
connection as well so if you have pointers on that that would be awesome!