iwanders/OBD9141 #52: None of the examples are working

fly2jerry, 2023-12-12:
Hi,
I tried to get any of the examples working, but no success at all.
I want to create a simulator on K-Line. I tried this with Arduino Uno and Mega.
I assume that with all the different libraries, I am not sure if I use the correct ones.
Do you have a zipped library with all needed files?

Regards
fly2jerry

fly2jerry, 2023-12-17:
Thank you for your response, but I do not even reach the status to follow any of #3 suggestions.
My issus is that I do not get it compiled. Even the simple one "reader". 
Thats why I asked for a set of libraries.

Here my error messages using Arduino UNO

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\flugl\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=2341_0043 -ide-version=10819 -build-path C:\Users\flugl\AppData\Local\Temp\arduino_build_595793 -warnings=none -build-cache C:\Users\flugl\AppData\Local\Temp\arduino_cache_89086 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\flugl\Documents\Arduino\OBD9141-master\examples\reader\reader.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\flugl\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=2341_0043 -ide-version=10819 -build-path C:\Users\flugl\AppData\Local\Temp\arduino_build_595793 -warnings=none -build-cache C:\Users\flugl\AppData\Local\Temp\arduino_cache_89086 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\flugl\Documents\Arduino\OBD9141-master\examples\reader\reader.ino
Using board 'uno' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\flugl\\AppData\\Local\\Temp\\arduino_build_595793\\sketch\\reader.ino.cpp" -o nul
Alternatives for WProgram.h: [OBD9141-master]
ResolveLibrary(WProgram.h)
  -> candidates: [OBD9141-master]
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\\h [... cut ...]
C:\Users\flugl\Documents\Arduino\libraries\OBD9141-master/TinyDebugSerial.h:686:22: error: cannot convert 'TinyDebugSerialWriter_38400* {aka int*}' to 'TinyDebugSerialWriter*' in assignment
           _writer = &tdsw38400;
                      ^~~~~~~~~
C:\Users\flugl\Documents\Arduino\libraries\OBD9141-master/TinyDebugSerial.h:690:22: error: cannot convert 'TinyDebugSerialWriter_115200* {aka int*}' to 'TinyDebugSerialWriter*' in assignment
           _writer = &tdsw115200;
                      ^~~~~~~~~~
Bibliothek OBD9141-master im Ordner: C:\Users\flugl\Documents\Arduino\libraries\OBD9141-master (legacy) wird verwendet
exit status 1
'Serial1' was not declared in this scope

iwanders, 2023-12-17:
Ah, well, ehm, I haven't touched an Arduino in years.

But I recall because the UNO only has one serial port you need a library to provide a software serial port. The [reader_softserial](https://github.com/iwanders/OBD9141/blob/master/examples/reader_softserial/reader_softserial.ino) example is applicable, it relies on the [AltSoftSerial](https://github.com/PaulStoffregen/AltSoftSerial) library. In your errors I see `TinyDebugSerial`, but I have no idea what that is. During the build it should automatically detect that it's an atmega 328p and go into [this if statement](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.h#L17-L20) to set the alt soft serial define and try to include it, if that doesn't happen, comment out line 13, such that [later](https://github.com/iwanders/OBD9141/blob/ff2d79538ec177c54da349f5d06b86f6b3bfd524/src/OBD9141.h#L90-L98) the correct serial type is used.