iwanders/OBD9141 #55: Usage with MCP2025

Dabi1127, 2024-01-14:
Hey!
Can I use this library with the MCP2025 IC? If so, do I need to change anything in the code?
MCP2025 datasheet: https://ww1.microchip.com/downloads/en/DeviceDoc/20002306B.pdf
I found it pretty similar to the SN65HVDA100.

iwanders, 2024-01-14:
Library is agnostic to the transceiver chip, it only uses digital signals on `Tx` and `Rx`, just make sure to set the wake pin appropriately, just glancing at the first page I expect it'll work fine. Give it a shot :)

supergladson, 2024-01-16:
Hello friends, speaking of working fine, I would like to share two new informations.  The first is that this library worked perfectly on the MC33660. The other information is that it worked on the STM32 family using the STM32Duino library. There are some considerations: 1) The user must insert the index number of the port array to be used in the obd.begin() function, as this function does not accept the PB9, PB10 format in RX_PIN, TX_PIN, only integers numbers. This information about the relationship between the array index and the pin name is in the variant_generic.cpp file. 2) To avoid communication problems on the STM32, configure the Serial port to SERIAL_8N1 when starting Serial.begin(), it is a good practice for the STM32Duino. To find out the default port of each STM32 microcontroller in the STM32DUINO library, open the variant_generic.h file that refers to the core of your microcontroller. To finish, in the STM32 (STM32DUINO) from the second serial port onwards, the initialization occurs using the HardwareSerial() function. As this OBD9141 library will use two serial ports, I will already provide this tip. I hope I helped how to get this OBD9141 library successful with the STM32 family using the STM32Duino library (Arduino_Core_STM32).