iwanders/OBD9141 #59: initKWP not completing initialization

FlippedBench, 2024-08-16:
Hello, I am attempting to use this library to communicate with a Denso ESC that uses fast KWP initialization. However, I have been unable to get the readerKWP example program to work. Probing it with an oscilloscope just shows a single pulse every time it attempts the initialization. 

Enabling the debug gets me the following readout:
Looping
Before 25 ms / 25 ms startup.
Enable port.
init_success:0

I am using an RP2040 based Arduino if that could cause any compatibility issues?

Thanks!

iwanders, 2024-08-17:
Single pulse makes sense, there's one pulse [at the start of initKWP](https://github.com/iwanders/OBD9141/blob/fc824827359c98ef6559d58080b51f23072f8834/src/OBD9141.cpp#L469-L481), but after that you should see the bytes listed there on line 481. If you don't see that, something is either wrong with the hardware, or with the serial port on your board.

Are you using the tx_pin that is associated to the correct serial port you are providing in the begin method?

FlippedBench, 2024-08-17:
I am assigning them to GPIO pins 10 and 11, which corresponds with how they are physically connected. I tried swapping them in case I mixed up RX and TX but no dice. When you specify what pins to use for the serial port, can it be any GPIO pins, or does it have to be UART TX and RX pins specifically?

iwanders, 2024-08-18:
> does it have to be UART TX and RX pins specifically?

Depends on your hardware, practically all hardware has only a few pins that can be used as a serial port though. The Tx and Rx passed to the begin function must match the Tx and Rx that are used for the serial port that is also passed in.