iwanders/OBD9141 #60: One ERRORR

toprakb66, 2024-10-04:
My friend, unfortunately, I couldn't get this to work on an Arduino Uno. It shows init 0. To be honest, I haven't seen a working example where someone reads data from the serial monitor using Arduino Uno. There are always tutorials on how it should be done, but I haven't seen a successful implementation with actual readings on the serial monitor. I've been working on this project for a month without any results. Can you help me?

toprakb66, 2024-10-04:
[image]
Hello my friend, thank you for your response. I am testing your code with the L9637D chip. My connection is like this. I have tried many codes and received the following response from the ECU: C1 33 F1 81 66. This is an error message, and I can't figure out why it is being generated. I've been working on this project for a month.

iwanders, 2024-10-04:
Did you try to google `C1 33 F1 81 66`, or search in the issues here?

A quick google finds me:
- [This document](https://www.obdclearinghouse.com/Files/viewFile?fileID=1380) which states that `C1 33 F1 81 66` is the `StartCommunication` request.
- [This page](https://evoecu.logic.net/wiki/OBD-II_Protocol) which states that is the package is sent out by the tester to the ECU
- The sequence also occurs in #14 

That sequence - in fact - is part of the [initKWP](https://github.com/iwanders/OBD9141/blob/fc824827359c98ef6559d58080b51f23072f8834/src/OBD9141.cpp#L481) handshake, it is not an error message, that sequence should be sent by the tester, to the ECU.

Is there another tester on the same bus? if so remove it.

If there's no other device, it could be that (for some reason) you aren't properly reading the echo from your own transmissions [in this function](https://github.com/iwanders/OBD9141/blob/fc824827359c98ef6559d58080b51f23072f8834/src/OBD9141.cpp#L62-L71), I've never seen that problem though, so I'd be surprised if that's the case. The debug prints may tell us whether this echo is seen in [this requestKWP](https://github.com/iwanders/OBD9141/blob/fc824827359c98ef6559d58080b51f23072f8834/src/OBD9141.cpp#L485C15-L485C25) call.

Please read the links, in particular the issue I linked and enable debug prints as I already stated in my first reply.

toprakb66, 2024-10-05:
"Hi, my friend. First of all, thank you for responding to my question. I am new to these things. I don't know how to add the code in [this link](https://github.com/iwanders/OBD9141/blob/master/src/OBD9141.h#L11) to Arduino. It gives me an error. Can you help me?"

toprakb66, 2024-10-05:
[image]
Why is no information appearing on the serial monitor?

toprakb66, 2024-10-05:
[image]
Buddy, I’m encountering this kind of error in another code. What do you think could be causing it?

iwanders, 2024-10-05:
> Why is no information appearing on the serial monitor?

I don't know, I haven't used the Arduino application in years, I'm sure there could be many reasons.

> Buddy, I’m encountering this kind of error in another code. What do you think could be causing it?

That should be the data you are sending to the ECU, like I said in [this comment](https://github.com/iwanders/OBD9141/issues/60#issuecomment-2394769800), please read comments carefully and follow the links provided.

The code in that screenshot doesn't look like it's my library, so I can't really help with that. That looks like [this one](https://github.com/muki01/OBD2_K-line_Reader/blob/72d2d5e52c39604f5bb6ebcc3805ae826ed34152/Basic_Code/K_Line.ino). You also made an issue in that repo in https://github.com/muki01/OBD2_K-line_Reader/issues/7 where @muki01 has been very patiently helping you out. In [this comment](https://github.com/muki01/OBD2_K-line_Reader/issues/7#issuecomment-2388669001) you obtained an `Init Success!!`, which I guess means things are working and you can correctly retrieved data?

If you got it working with one library, you can use a logic analyzer or second Arduino to record what happens on the K-line bus when it works, and compare to what happens on the bus for the library that doesn't work.

But if I were you I'd just go with the library that works, if you got it working with @muki01's library, that's great, I recommend sticking to using what works for you, especially if you are new to all this.

muki01, 2024-10-05:
Hello.  It gets such data because in my code I don't remove the codes that the microcontroller sends.  I know they are supposed to be deleted but it was slowing down the data receiving a bit and I decided not to delete them for now.  His problem is that after making the pattern and getting 55 08 08 and sending the reverse KW2 he doesn't get a response from the car.  Maybe my code has a problem but I have tested with several cars with the same protocol and it works fine for me.

toprakb66, 2024-10-06:
I think I'm using the ISO14230 protocol; when I connected the Wi-Fi OBD2 device to the vehicle, it communicated directly using this protocol. @muki01