iwanders/OBD9141 #62: Fiat ECU Marelli IAW - 4800bps

jorlandobr, 2025-03-23:
iwanders, congratulations for your project, very nice!

I am insterested in using/modifying for my car's ECU, a FIAT using a Marelli IAW ECU.

Main differences from your project:

- 4800bps (easy... modify #define OBD9141_KLINE_BAUD)
- magic 5 baud wake-up address 0x10h. Easy, modify lines doing the bitbang at TX pin
- your project K line is used for ECU startup and communication with ECU (ISO 5). IAW uses L line for ECU startup and K line for communication (ISO 8). Easy also, a diode in parallel with K and L lines will echo K to L, waking up the ECU. Echoes on L during communication don't interfere
- after startup your project has a delay up to 300ms to get an answer from the ECU or the proccess restarts. Here I got stuck. My ECU can take up to 2s to answer the wake-up signal (around 1.3s typical). I couldn't figure how to modify this delay (at least from what I can observe)

I tried both init and initKWPslow, none worked.

As you can see from the analyser capture, I have initialization successful, 1.28s after request:

[image]

And the response is correct for my ECU, 0x55b086831523:

[image]

But from my logs, almos immediattely (After setting port/First read is) the program tries to identify byte 0x55 (I tryed to print all the bytes from the buffer to see if it was being filled with the correct bytes at some point):

`[0 00:00:00.062] Looping
[0 00:00:00.093] Before magic 5 baud. Fiat 0x10
[0 00:00:00.093] Before setting port.
[0 00:00:00.093] After setting port.
[0 00:00:00.093] First read is: 0
[0 00:00:00.093] 10
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 255
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 255
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 152
[0 00:00:00.093] 23
[0 00:00:00.093] 252
[0 00:00:00.093] init_success:0
[0 00:00:00.093] Looping
[0 00:00:00.093] Before magic 5 baud. Fiat 0x10
[0 00:00:00.093] Before setting port.
[0 00:00:00.093] After setting port.
[0 00:00:00.093] First read is: 0
[0 00:00:00.093] 10
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 0
[0 00:00:00.093] 255
`

Do you hane any suggestion, given this scenario?

iwanders, 2025-03-23:
Hey, thanks for the detailed writeup.

> after startup your project has a delay up to 300ms to get an answer from the ECU or the proccess restarts. Here I got stuck. My ECU can take up to 2s to answer the wake-up signal (around 1.3s typical). I couldn't figure how to modify this delay (at least from what I can observe)

That delay is handled with the timeout of the serial bus, it's currently hardcoded:
https://github.com/iwanders/OBD9141/blob/fc824827359c98ef6559d58080b51f23072f8834/src/OBD9141.cpp#L379

I'd say start with changing that and making it more tolerant of your ECU?

jorlandobr, 2025-03-31:
Thanks for your reply!

I think that my problem may be related to the hardware serial initialization on ESP32. I was getting more success (initialization, sending and receiving acks, etc) with the code from, I since my K line circuit is the same for both, I started looking for initialization of serial ports on both projects.

My knowledge in C and C++ language is very basic, but from what I could understand from your code from OBD9141.h
```
    public:
        OBD9141();

        void begin(OBD_SERIAL_DATA_TYPE & serial_port, uint8_t rx_pin, uint8_t tx_pin);
        // begin method which allows setting the serial port and pins.
```

And from OBD9141.cpp:
```
  this->serial->begin(OBD9141_KLINE_BAUD, OBD9141_KLINE_TYPE);
```

So I believe that the resulting configuration line would the equivalent to

**Serial1.begin(4800, RX_PIN, TX_PIN);**

That,s ok, according the serial.begin page at [arduino docs,](https://docs.arduino.cc/language-reference/pt/en/functions/communication/serial/begin/) serial **config** is optional, 8N1 is assumed if not given, but I couldn't find the same affirmation for the ESP32's extra hardware ports.

So I made a small sketch to test this, and if config wasn't used, the serial wasn't initialized,  as can be seen on the capture from the analyzer. The low vales between communications should have communication without the config parameter, accordingly

[image]

[image]

I tried a small moddification to OBD9141.h adding
```
#define OBD9141_KLINE_TYPE SERIAL_8N1
```
And changing OBD9141.cpp 
```
this->serial->begin(OBD9141_KLINE_BAUD, OBD9141_KLINE_TYPE);
```
but still not geting the answer from the ECU. But since I don't know that much, maybe I missed (or messed) something.

My test sketch (I also tested beggining using the config, same results. Only have communication when the config parameter is used):

```
#include "Arduino.h"

#define RX_PIN 16 // connect to transceiver Rx for ESP32
#define TX_PIN 17 // connect to transceiver Tx

void setup(){
    Serial1.begin(4800, RX_PIN, TX_PIN);
    //Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
    Serial1.print("init: Serial1.begin(4800, RX_PIN, TX_PIN);");
    delay(630);
    Serial1.end();

    Serial1.begin(4800, SERIAL_8N1, RX_PIN, TX_PIN);
    //Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
    Serial1.print("init: Serial1.begin(4800, SERIAL_8N1, RX_PIN, TX_PIN);");
    delay(630);
    Serial1.end();

    Serial1.begin(4800, RX_PIN, TX_PIN);
    //Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
    Serial1.print("init: Serial1.begin(4800, RX_PIN, TX_PIN);");
    delay(630);
    Serial1.end();

    Serial1.begin(4800, SERIAL_8N1, RX_PIN, TX_PIN);
    //Serial2.begin(115200, SERIAL_8N1, RXD2, TXD2);
    Serial1.print("init: Serial1.begin(4800, SERIAL_8N1, RX_PIN, TX_PIN);");
    delay(630);
    Serial1.end();

    delay(500);
}

void loop(){

delay(100);

}
```

iwanders, 2025-04-01:
Interesting, this library predates the ESP32 by years, in #45  it was pretty much the same situation.

Given that you state that [`muki01/OBD2_K-line_reader`](https://github.com/muki01/OBD2_K-line_Reader/) does work... we should be able to compare the init from there:

jorlandobr, 2025-04-02:
I did a few hours of testing today, trying to be more precise on the problem and maybe solving it.

I messed a lot on your original code and was getting lost, so I started from your clean code and tried to find the minimum to get replies from my ECU.

So, hard coding the replies I can wake-up the ECU (0x33 seems to be standard, mine uses 0x10), then I could send a select and get an ack, send a select host and get a set host answer and, at last, send a request the design identification and get answer.

To get to it I had to initialize the serial on the void setup routine. 

```cpp
#include <Arduino.h>
#include "OBD9141.h"

// ESP32 example, courtesy of gauix4; https://github.com/iwanders/OBD9141/issues/38

#define RX_PIN 16 // connect to transceiver Rx for ESP32
#define TX_PIN 17 // connect to transceiver Tx

bool init_success;

OBD9141 obd;

void setup(){
    Serial.begin(115200);
    Serial1.begin(4800, SERIAL_8N1, RX_PIN, TX_PIN); // Needed to start hardware serial 1 here
    //Serial1.println("initialization");
    obd.begin(Serial1, RX_PIN, TX_PIN);

    pinMode(19, OUTPUT); // pin 19 works as L line
    digitalWrite(19, LOW);  // just need to go up during slow init.

    delay(500);
    Serial.println("initialization");
    Serial.println("");
  
    while (!obd.init()){
        return;
    }
}

```

Also,I had to comment out the set_port(false);. After the set_port disabled even my modified line would't make the TX come back to activity when the code gets to set_port(true); a few lines below...

```cpp
bool OBD9141::initImpl(bool check_v1_v2){
    use_kwp_ = false;
    // this function performs the ISO9141 5-baud 'slow' init.
    //this->set_port(false); // disable the port. **** comment this line or port won't come back
    this->kline(true);
    delay(OBD9141_INIT_IDLE_BUS_BEFORE); // no traffic on bus for 3 seconds.
    OBD9141println("Before magic 5 baud.");
    // next, send the startup 5 baud init..
    //this->kline(false); delay(200); // start
    //this->kline(true); delay(400);  // first two bits
    //this->kline(false); delay(400); // second pair
    //this->kline(true); delay(400);  // third pair
    //this->kline(false); delay(400); // last pair
    //this->kline(true); delay(200);  // stop bit
	
    digitalWrite(19, HIGH), delay(200);
    digitalWrite(19, HIGH), delay(400);
    digitalWrite(19, HIGH), delay(400);
    digitalWrite(19, LOW), delay(200);
    digitalWrite(19, HIGH), delay(400);
    digitalWrite(19, HIGH), delay(200);
    digitalWrite(19, LOW), delay(200);

    // this last 200 ms delay could also be put in the setTimeout below.
    // But the spec says we have a stop bit.

[... 75 lines cut ...]
    digitalWrite(19, LOW), delay(200);
    digitalWrite(19, HIGH), delay(400);
    digitalWrite(19, HIGH), delay(200);
    digitalWrite(19, LOW), delay(200);

    // this last 200 ms delay could also be put in the setTimeout below.
    // But the spec says we have a stop bit.

    // done, from now on it the bus can be treated ad a 10400 baud serial port.

    OBD9141println("Before setting port.");
    //this->set_port(true); *** does not work
	//this->serial->begin(OBD9141_KLINE_BAUD, OBD9141_KLINE_TYPE); *** does not work
	Serial1.begin(4800, SERIAL_8N1, 16, 17); //, RX_PIN, TX_PIN); // worked 
    OBD9141println("After setting port.");
    uint8_t buffer[1];
    OBD9141println("Before timeout.");
	
	delay(1300);

```

Makes any sense to you, not working calling from your library but works if I use the full command?

Thanks for the cpp tip :-)

iwanders, 2025-04-02:
Thanks for doing a detailed and systematic approach.

Sounds like the problems do indeed stem from that Serial initialisation, which just needs to be called slightly differently on the ESP32? The if statement from my comment [here](https://github.com/iwanders/OBD9141/issues/62#issuecomment-2770821932) may just do the trick?

I've pushed the [issue-62-wip](https://github.com/iwanders/OBD9141/tree/issue-62-wip) branch, which has this diff: https://github.com/iwanders/OBD9141/compare/master...issue-62-wip

So effectively it changes the begin call in `set_port` to be the one required by the ESP. Could you test if just using that branch works?

jorlandobr, 2025-04-03:
This does not work:

```cpp
        // Esp32 needs the full being command with type and pins to work.
        #if ESP32
```

```cpp
void OBD9141::set_port(bool enabled){
	OBD9141println("// OBD9141::set_port(bool enabled){ - line 23 ");
    if (enabled){
        OBD9141println("// Work around the incorrect pinmode configuration in Due. line 25 ");
        #ifdef ARDUINO_SAM_DUE
		  OBD9141println("#ifdef ARDUINO_SAM_DUE - LINHA 27");
          g_APinDescription[this->rx_pin].pPort -> PIO_PDR = g_APinDescription[this->rx_pin].ulPin;
          g_APinDescription[this->tx_pin].pPort -> PIO_PDR = g_APinDescription[this->tx_pin].ulPin;
        #endif
		
		// Esp32 needs the full being command with type and pins to work.
        #if ESP32
        //#ifdef ESP32
		  OBD9141println("// Esp32 needs the full being command with type and pins to work. - line 32 ");
          this->serial->begin(OBD9141_KLINE_BAUD, SERIAL_8N1, this->rx_pin, this->tx_pin);
        #else
          this->serial->begin(OBD9141_KLINE_BAUD);
		  OBD9141println("// How did I got here??? ");		  
        #endif
```
My console output. See that it goes to the else directive:

```
Before magic 5 baud.
Before setting port.
// OBD9141::set_port(bool enabled){ - line 23
// Work around the incorrect pinmode configuration in Due. line 25
**// How did I got here???**
After setting port.
Before timeout.
```

This works:

```cpp
        // Esp32 needs the full being command with type and pins to work.
        #ifdef ESP32
```

```cpp
void OBD9141::set_port(bool enabled){
	OBD9141println("// OBD9141::set_port(bool enabled){ - line 23 ");
    if (enabled){
        OBD9141println("// Work around the incorrect pinmode configuration in Due. line 25 ");
        #ifdef ARDUINO_SAM_DUE
		  OBD9141println("#ifdef ARDUINO_SAM_DUE - LINHA 27");
          g_APinDescription[this->rx_pin].pPort -> PIO_PDR = g_APinDescription[this->rx_pin].ulPin;
          g_APinDescription[this->tx_pin].pPort -> PIO_PDR = g_APinDescription[this->tx_pin].ulPin;
        #endif
		
		// Esp32 needs the full being command with type and pins to work.
        //#if ESP32
        #ifdef ESP32
		  OBD9141println("// Esp32 needs the full being command with type and pins to work. - line 32 ");
          this->serial->begin(OBD9141_KLINE_BAUD, SERIAL_8N1, this->rx_pin, this->tx_pin);
        #else
          this->serial->begin(OBD9141_KLINE_BAUD);
		  OBD9141println("// How did I got here??? ");		  
        #endif
```

My console output:

```
Before magic 5 baud.
Before setting port.
// OBD9141::set_port(bool enabled){ - line 23
// Work around the incorrect pinmode configuration in Due. line 25
**// Esp32 needs the full being command with type and pins to work. - line 32**
After setting port.
Before timeout.
```

iwanders, 2025-04-03:
Ah, yes of course, thanks for catching that, that should've been `#if ESP32==1` or `#ifdef`.

I pushed an update to the branch; https://github.com/iwanders/OBD9141/tree/issue-62-wip

No additional changes necessary to make it work with that fix for `begin()`? Because if this works I'm going to merge it into `master`... :)

jorlandobr, 2025-04-04:
These updates made begin() work. All the modifications that I plan to do are for the particularites of my ECU.

I checked keeping the code from the site, making just minimal modifications to get answers from my ECU.

I think that ESP32 users won't have weird problems anymore. :-)

iwanders, 2025-04-04:
I'll take it!

I've filed https://github.com/iwanders/OBD9141/pull/63, squashed the commits and added you to the commit with `Co-Authored-By`, I took your email & name from a commit in `ntc-templates` to ensure it ties to your github user. Let me know if you're ok with how it looks, I'll merge it only then.

jorlandobr, 2025-04-04:
Seem fine to me :-)

Glad to help :-)

iwanders, 2025-04-05:
Thanks so much for your report, testing and debugging @jorlandobr, I've merged the PR into master and also commented on the other related issues that they're likely resolved now.

closed 2025-04-05 by iwanders